RaspAP/raspap-webgui #1325: [Security] OS Command injection vulnerability in configure_client.php

eldstal, 2023-03-29:
An authenticated user is able to inject OS commands via POST parameters to /wpa_conf here:

https://github.com/RaspAP/raspap-webgui/blob/592bce02bfee1169ca92e5e93ebded66bac42c6e/includes/configure_client.php#L20

The proper fix is probably similar to yesterday's issue.

Thanks again for the quick response!

eldstal, 2023-03-30:
Since the old report wasn't intended to be removed, I've updated this issue to serve as a report for all three similar vulnerabilities.

Note: The two mentioned vulnerabilities in hostapd.php have already been fixed in [PR 1322](https://github.com/RaspAP/raspap-webgui/pull/1322).

billz, 2023-04-01:
Resolved with #1326, @eldstal thanks for reporting and the PR

closed 2023-04-01 by billz