RaspAP/raspap-webgui #1336: Installation problem after apt upgrade

Toadie1, 2023-04-09:
## Checklist

- [ x] This is a bug report
- [ x] I observed this bug on a clean install of the OS
- [ x] I have followed the project prerequisites
- [ x] I have searched this repository for existing issues
- [ x] I checked the FAQ and official documentation before creating this issue
- [ x] I have read and understand the issue reporting guidelines 

## Bug description

## Your environment
1. Operating System: RPi OS 32-bit Lite 
2. Hardware and version: RPi 4 (4GB)
3. RaspAP version: 2.8.9
4. Clean install of a compatible operating system? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? No
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? No

## Steps to reproduce
I have a clean installation of Raspberry OS 32bit right now created via Raspberry Pi Imager.

After the first boot I execute the routine 
```sh
sudo apt update
sudo apt upgrade
sudo reboot now
```
After the boot, the system is switched to 64bit kernel.
So I add the `arm_64bit=0` parameter to `/boot/config.txt` and reboot.

After boot I have version `Linux travel-cube 6.1.21-v7l+ #1642 SMP Mon Apr 3 17:22:30 BST 2023 armv7l`

I run the RaspAP installation with `sudo curl -sL https://install.raspap.com | bash` and then the installation crashes.
Here is listing of the console output...
```console
toadie@travel-cube:~ $ uname -a
Linux travel-cube 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux
toadie@travel-cube:~ $ sudo curl -sL https://install.raspap.com | bash

 888888ba                              .d888888   888888ba
 88     8b                            d8     88   88     8b
a88aaaa8P' .d8888b. .d8888b. 88d888b. 88aaaaa88a a88aaaa8P
 88    8b. 88    88 Y8ooooo. 88    88 88     88   88
 88     88 88.  .88       88 88.  .88 88     88   88
 dP     dP  88888P8  88888P  88Y888P  88     88   dP
                             88
                             dP      version 2.8.9

The Quick Installer will guide you through a few easy steps

RaspAP Install: Configure installation
Detected OS: Raspbian GNU/Linux 11 (bullseye)
Using GitHub repository: RaspAP/raspap-webgui 2.8.9 branch
Configuration directory: /etc/raspap
lighttpd root: /var/www/html? [Y/n]: y
Installing lighttpd directory: /var/www/html
Complete installation with these values? [Y/n]: y
RaspAP Install: Updating sources
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Reading package lists...
RaspAP Install: Checking for systemd network services
systemd-networkd.service is not running (ok)
systemd-resolved.service is not running (ok)
[ ✓ ok]
RaspAP Install: Installing required packages
[... 270 lines cut ...]
sed: can't read /var/www/html/includes/config.php: No such file or directory
[ ✘ error] Unable to modify config.php
[ ✓ ok]
RaspAP Install: Adding raspap.sudoers to /etc/sudoers.d/090_raspap
cp: cannot stat '/var/www/html/installers/raspap.sudoers': No such file or directory
[ ✘ error] Unable to apply raspap.sudoers to /etc/sudoers.d/090_raspap
chmod: cannot access '/etc/sudoers.d/090_raspap': No such file or directory
[ ✘ error] Unable to change file permissions for /etc/sudoers.d/090_raspap
RaspAP Install: Symlinked wpa_supplicant hooks for multiple wlan interfaces
RaspAP Install: Unmasking and enabling hostapd service
Unit hostapd.service does not exist, proceeding anyway.
Failed to enable unit: Unit file hostapd.service does not exist.
[ ✓ ok]
RaspAP Install: Installation completed
Join RaspAP Insiders for early access to exclusive features!

> https://docs.raspap.com/insiders/
> https://github.com/sponsors/RaspAP/

toadie@travel-cube:~ $
```

What am I doing wrong? 
Alternatively, am I the first to try to install RaspAP on the new Raspberry Pi OS kernel released on April 3, 2023?
Thank you

b0hne, 2023-04-09:
same here, was about to open this topic

closed 2023-04-11 by billz

tegeek1445, 2023-04-11:
++ Even I have the same issue

qirpi, 2023-04-11:
Same issue here --  Is there a fix for this? Why is this closed?

volisoft, 2024-06-03:
Same issue with the latest installer and Raspberry PI OS Lite.

PR #1337 (2023-04-11): Fix: update _install_dependencies()
Resolves #1336