RaspAP/raspap-webgui #1352: Installing telnetd hangs in "sudo apt install telnetd"

2bholland, 2023-06-01:
## Checklist

- [ ] This is a bug report
- [ ] I observed this bug on a clean install of the OS
- [ ] I have followed the project prerequisites
- [ ] I have searched this repository for existing issues
- [ ] I checked the FAQ and official documentation before creating this issue
- [ ] I have read and understand the issue reporting guidelines 

## Bug description

The telnetd application can not be installed on my fresh install.

Raspberry Pi OS Lite (Legacy)

    Release date: May 3rd 2023
    System: 32-bit
    Kernel version: 5.10
    Debian version: 10 (buster)
    Size: 321MB

Tested on Raspberry Pi 2B and 3B fresh OS installs with multiple SD cards of different brands and sizes.  The app partially installs then hangs and refuses to continue.  If I break the install with ^C then try again i get the error telling me that there was a a failed install and I must continue installing the failed install... But the failed install will not continue.  The only way to recover is to re-format the SD card and re-install the OS.  It appears the downloaded file is incomplete or damaged.

Where the install hangs is here:

Preparing to unpack .../telnetd_0.17-41.2_armhf.deb ...
Unpacking telnetd (0.17-41.2) ...
Setting up update-inetd (4.49) ...
Setting up tcpd (7.6.q-28) ...
Setting up openbsd-inetd (0.20160825-4) ...
client_loop: send disconnect: Broken pipe
bill@z-77x2:~$ 
Progress: [ 76%] [###########################################################################........................] 

This is not recoverable.  The Pi can be re-started but nothing else can ever be installed because of this error.

After powering off the pi and rebooting, then trying the install again I get this:

bill@raspberrypi2:~ $ sudo apt install telnetd
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem. 
bill@raspberrypi2:~ $ sudo dpkg --configure -a
Setting up openbsd-inetd (0.20160825-4) ...

Progress: [ 76%] [###########################################################################........................] 

And now it is hung again at the same place.

If there is anything that is unclear or that I have left out, just try to install telnetd on the Pi version of Debian Buster and it will be clear.

If I have messed this up please forgive me as this is my first attempt at reporting a problem here and it is a little daunting for a beginner.

## Your environment
1. Operating System: **ENTER HERE** 

     Tried on Raspberry Pi 2B and 3B

6. RaspAP version: 
7. Clean install of a compatible operating system? 

  This was tried on several clean installs with SD cards of various sizes from different manufacturers.

8. RaspAP Quick Install or Manual setup? 

  SD cards created with Raspberry Pi Imager.

 9. Using default configuration? 

  Using default install configuration, yes.

11. Simultaneous AP and managed mode? 
12. Onboard wireless chipset or external adapter? 

  Installs attempted using both an Ethernet terminal and a monitor and keyboard.

13. Other software or services running with RaspAP? 

  a fresh boot and nothing else was running.

## Steps to reproduce

  Easy to reproduce, just try to install telnetd.

## Screenshots

## Additional context

billz, 2023-06-02:
Duplicate of #1351

billz, 2023-06-02:
There are a few things going on here, unrelated to your RaspAP install, but let's look at them:

1. You're trying to install the telnet server `telnetd` - the client is `telnet`. Are you sure you need telnet server?
2. telnet is generally considered obsolete and running a telnet server on your Pi is a security risk. ssh is much newer and more secure than telnet.
3. inetd is manipulating your network interface. It's being taken down temporarily which results in the hanging session and broken pipe. Again, if you really need telnetd try installing it with an attached monitor - not ssh.
4. If you want to debug the package manager, run it with no hang-up and redirect the output to a somewhere you can view the log: `nohup sudo dpkg --configure -a > $HOME/dpkg.log`

closed 2023-06-08 by billz