RaspAP/raspap-webgui #1353: Add support for Debian/Armbian Bookworm

littlegraycells, 2023-06-01:
## Is your feature request related to a problem? 
I would like to upgrade my base OS to the newest LTS release of Debian 12 (Bookworm).

## Describe the solution you'd like
It would be great if RaspAP could support Debian/Armbian Bookworm for new installs. 

## Additional context
Debian 12 Bookworm is currently in the testing phase but expected to release on June 10, 2023. The current install scripts fails when trying to install on this version

`Armbian 23.05.1 bookworm is unsupported. Please install on a supported distro.`

billz, 2023-06-02:
Debian 12 (bookworm) and Armbian 23.05 (Suni) are separate distros. Both are on the project roadmap.
Can you assist with testing?

littlegraycells, 2023-06-02:
Yes, happy to. I can test Debian on an x86 machine. For Armbian, I have an Odroid C2 I can test on.

billz, 2023-06-03:
@littlegraycells great. Have a look at #1354. So far I have a stable install with Debian 12 on a Pi 3B+. There are two small changes that need to be made for php8.2-cgi. 

For the installer, you can edit [raspbian.sh](https://github.com/RaspAP/raspap-webgui/blob/master/installers/raspbian.sh) and force it to load a local `common.sh` like so:

```
else
    source="common"
    component="Install"
    source /home/${source}.sh
    _install_raspap || _install_status 1 "Unable to install RaspAP"
fi
```

Substitute `/home` for the location of common.sh on your device.

littlegraycells, 2023-06-03:
Okay so it's installed fine and seems stable right now on Armbian 23.05.1 on an Odroid C2. Will report back after a few days.

billz, 2023-06-04:
That's very helpful, thanks. My Armbian test devices are Orange Pis that [aren't booting with bookworm](https://forum.armbian.com/topic/28625-armbian-jammy-235-and-bookworm-latest-versions-stuck-on-boot/#comment-166034) at the moment.

closed 2023-08-01 by billz