RaspAP/raspap-webgui #1356: Sponsor: Speedtest-Server in the EU

Exanteros, 2023-06-09:
Hey :)
As I am building my RaspAP setup I came across the speedtest server of RaspAP. Unfortunately there is only one location I can test my internet speed to, thats in the US. 

I would love to sponsor a location inside the EU, if that's possible.

Thanks.

billz, 2023-06-10:
Hey there! That would be great.

Start by cloning our Speedtest repo and host it with the web service of your choice:
https://github.com/RaspAP/speedtest/

We use nginx but any reasonably fast web server will do. You can of course update the sponsor link as you wish. When this is done, we'll update the [list of available servers](https://github.com/RaspAP/speedtest/blob/12cef7b91b75e9c96552a4dd0a13f7047ce6933d/index.html#L37) and map a subdomain to your IP. Give a shout if you need anything.

Thanks :)

Exanteros, 2023-06-15:
I am currently working on the speedtest server. Looking forward on finishing it next week.

closed 2023-12-12 by github-actions[bot]