RaspAP/raspap-webgui #1366: Unable to change DHCP-Tab sub categories

xXPerditorXx, 2023-07-08:
## Checklist

- [x] This is a bug report
- [x] I observed this bug on a clean install of the OS
- [x] I have followed the project prerequisites
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

If you go to the DHCP Server Tab, I'm not able to change to Client-List, Loggin or any of the other things. It just sets #loggin or #client-list in the url but nothing is changing.
## Your environment
1. Operating System: **Debian GNU/Linux 11 (bullseye) aarch64 (Raspberry Pi OS)** 
2. Hardware and version: Raspberry Pi 4 Model B Rev 1.4 8GB
3. RaspAP version: 2.9.2
4. Clean install of a compatible operating system? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? No
7. Simultaneous AP and managed mode? wlan0 is the hotspot, Wifi over eth0 so I think yes?
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? 
Apache in folder /var/www/html/subwebsite on port 8080
## Steps to reproduce

If you go to the DHCP Server-Tab and tap on Client-List or Loggin etc. the #logging gets added to the url but the subcategory isn't changing.

## Screenshots

https://github.com/RaspAP/raspap-webgui/assets/60017046/183afb83-0eb3-40c7-b08c-cf559a691e7e

## Additional context

The problem was already in 2.8.5. Tried it with Firefox, Safari (iPadOS), Chromium

billz, 2023-07-08:
https://github.com/RaspAP/raspap-webgui/assets/229399/6457be87-3363-4b69-ab29-3637baa316ae

Checked with:
Safari Version 16.2 (18614.3.7.1.5)
Brave Version 1.52.129 Chromium: 114.0.5735.198 (Official Build) (arm64)

Can't really advise you beyond inspecting with developer tools.

xXPerditorXx, 2023-07-08:
After full reinstall, everything seems fine.

xXPerditorXx, 2023-07-08:
So, after a bit playing around, I noticed, if you enable dhcp request loggin aswell as dns logging, that this problem is occuring.

billz, 2023-07-08:
Aha, that's a good clue. DHCP and DNS logging can create very big log files, if not cleared periodically. Tab switching is most likely delayed because the UI must reload many MB of log data to populate the logging tab. Clearing the log should help.

xXPerditorXx, 2023-07-09:
The thing is, that there shouldn't be anything inside the logs. I just acivat loggin and it instantly breaks everything which is inside the DHCP Server Tab.

Wouldn't it be possible, to add a feature that limits those two log files to a specific size the user sets?

billz, 2023-07-09:
It's a single system log file located at `/var/log/dnsmasq.log`. You can check the contents of this file from the shell and clear it manually with `truncate -s 0 [file]` or `cat /dev/null > [file]`. 

> add a feature that limits those two log files to a specific size the user sets

Yes, that's a good idea. The System tab would be the best place for this setting.

billz, 2023-11-28:
@xXPerditorXx thanks for raising this. I'd like to give you access to Insiders as a token of gratitude, just say the word

PR #1465 (2023-11-29): Feature: Set diagnostic log size limit
Resolves #1366
closed 2023-11-29 by billz