RaspAP/raspap-webgui #1374: Bridged AP mode grayed out

Title666645, 2023-08-03:
I'm unable to toggle Bridged AP mode as it is grayed out

## Checklist

- [x] This is a bug report
- [x] I observed this bug on a clean install of the OS
- [x] I have followed the project prerequisites
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

I am trying to toggle Bridged AP mode, but it is inaccessible. I reinstalled RaspAP and the operating system, but it is still grayed out.

## Your environment
1. Operating System: Armbian Jammy 
2. Hardware and version: RPi 4 8gb
3. RaspAP version: 2.9.3
4. Clean install of a compatible operating system? Yes
5. RaspAP Quick Install or Manual setup? Quick install
6. Using default configuration? Yes 
7. Simultaneous AP and managed mode? Yes 
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? Samba

## Steps to reproduce

Go to Hotspot > Advanced, and see grayed out Bridged AP mode

## Screenshots

[image]

## Additional context

billz, 2023-08-03:
Execute `lsb_release -sd` in the shell and post the result.

Title666645, 2023-08-03:
Ubuntu 22.04.2 LTS

billz, 2023-08-03:
https://docs.raspap.com/bridged/#limitations

Title666645, 2023-08-03:
Ohhh, sorry for bothering.

closed 2023-08-03 by Title666645