RaspAP/raspap-webgui #1382: insider access - github password authentication fails

thetravellor, 2023-08-19:
## Checklist

- [ x] This is a bug report
- [ x] I observed this bug on a clean install of the OS
- [ x] I have followed the project prerequisites
- [ x] I have searched this repository for existing issues
- [ x] I checked the FAQ and official documentation before creating this issue
- [ x] I have read and understand the issue reporting guidelines 

## Bug description

```
Username for 'https://github.com': thetravellor
Password for 'https://thetravellor@github.com':
remote: Support for password authentication was removed on August 13, 2021.
remote: Please see https://docs.github.com/en/get-started/getting-started-with-git/about-remote-repositories#cloning-with-https-urls for information on currently recommended modes of authentication.
fatal: Authentication failed for 'https://github.com/RaspAP/raspap-insiders/'
```

also fails with email address

```
raspAP Install: Cloning latest files from github
Cloning into '/tmp/raspap-webgui'...
Username for 'https://github.com': cooperstephenjames@gmail.com
Password for 'https://cooperstephenjames@gmail.com@github.com':
remote: Support for password authentication was removed on August 13, 2021.
remote: Please see https://docs.github.com/en/get-started/getting-started-with-git/about-remote-repositories#cloning-with-https-urls for information on currently recommended modes of authentication.
fatal: Authentication failed for 'https://github.com/RaspAP/raspap-insiders/'
[ ✘ error ]  Unable to download files from github
mv: cannot stat '/tmp/raspap-webgui': No such file or directory
```

completely removes working environment and renders AP completely inoperable.

## Your environment
1. Operating System: Raspberry Pi OS 64-bit lite
3. Hardware and version:  Raspberry P 4b 8GB
5. RaspAP version: 2.9.1 I think
7. Clean install of a compatible operating system? Yes
8. RaspAP Quick Install or Manual setup? Quick install
10. Using default configuration? No, I edited in AU as a supported Wireless 5g AC country.
12. Simultaneous AP and managed mode? Not sure
14. Onboard wireless chipset or external adapter? Onboard
16. Other software or services running with RaspAP? No

## Steps to reproduce
Attempt to run curl -sL https://install.raspap.com | bash -s -- --upgrade --insiders

## Screenshots
see log file
[raspap_install.log](https://github.com/RaspAP/raspap-webgui/files/12386069/raspap_install.log)
## Additional context

billz, 2023-08-19:
https://docs.raspap.com/insiders/#authentication

Discussed in [152](https://github.com/orgs/RaspAP/discussions/152#discussioncomment-5270136), [87](https://github.com/orgs/RaspAP/discussions/87) and [150](https://github.com/orgs/RaspAP/discussions/150).

closed 2023-08-19 by billz

thetravellor, 2023-08-19:
My word what a frustrating experience.

This requires GCM which does not either have a binary deb for raspberry pi, or successfully compile from source.

billz, 2023-08-19:
This is imposed by GitHub, unfortunately we have no control over it. 
Alternatively, you can upload a public key to GitHub. This has worked fine in the past.

thetravellor, 2023-08-19:
I have no idea how to use GPG keys with this script, although I do have one stored on git hub. do you a link to what you are referring?

Ok, moved forward a little. I have worked out how to pass a token using your script.

```
aspAP Install: Optimize PHP configuration
Enable HttpOnly for session cookies (Recommended)? [Y/n]:
Php-cgi enabling session.cookie_httponly.
RaspAP Install: Cloning latest files from github
Cloning into '/tmp/raspap-webgui'...
Username for 'https://github.com': thetravellor
Password for 'https://thetravellor@github.com':
remote: Write access to repository not granted.
fatal: unable to access 'https://github.com/RaspAP/raspap-insiders/': The requested URL returned error: 403
```

This a permissions error on yor end it seems, please can you have a look?

thetravellor, 2023-08-19:
stephencooper@raspap:~ $ ssh -T git@github.com
The authenticity of host 'github.com (20.248.137.48)' can't be established.
ECDSA key fingerprint is SHA256:p2QAMXNIC1TJYWeIOttrVc98/R1BUFWu3/LiyKgUfQM.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'github.com,20.248.137.48' (ECDSA) to the list of known hosts.
Hi thetravellor! You've successfully authenticated, but GitHub does not provide shell access.

thetravellor, 2023-08-19:
From the github website:

"If you authenticate without GitHub CLI, you must authenticate with a personal access token. When Git prompts you for your password, enter your personal access token"

I am doing this, it is failing

I have both gpg and ssh keys in place and published to github. 

This is a blocking issue and unresolved.

thetravellor, 2023-08-19:
Ok. The solution was as follows

1. Install the gh cli from github's repo.
2. Generate a classic personal token, as the newer fine grained tokens do not work.
3. Run "gh auth login", use the https method to log in when prompted, supply the recently generated classic token
4. Run your script, with the token as a parameter

Its up and running....but that was way too hard

billz, 2023-08-19:
GitHub tokens can be a pain to setup if not done before. I've found the method [described here](https://github.com/orgs/community/discussions/25259#discussioncomment-3247158) to be useful for troubleshooting repo access with classic personal tokens.

thetravellor, 2023-08-19:
Spoke to soon,

Now I get 404 Not Found on every page except dashboard

thetravellor, 2023-08-19:
Ill just nuke the sdcard I think.

thetravellor, 2023-08-20:
Ok, its now all working.

billz, 2023-08-23:
@thetravellor re: 5 GHz capabilities and AU, please see this FAQ
https://docs.raspap.com/faq/#wificountries

It's possible the wireless reg db was updated in the latest kernel release. PR's are welcome 😉