RaspAP/raspap-webgui #1389: Cannot add wifi network with spaces in it's name

darnokb, 2023-09-04:
## Checklist

- [x] This is a bug report
- [x] I observed this bug on a clean install of the OS
- [x] I have followed the project prerequisites
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines 

## Bug description

When trying to connect to wifi network with spaces in it's name RaspAP reports following error:

Passphrase must be 8..63 characters

## Your environment
1. Operating System: *RPi OS 32-bit lite 
2. Hardware and version: RPi Zero W 2 
3. RaspAP version: 2.9.3 
4. Clean install of a compatible operating system? Yes 
5. RaspAP Quick Install or Manual setup? Quick Install 
6. Using default configuration? Yes 
7. Simultaneous AP and managed mode? Yes 
8. Onboard wireless chipset or external adapter? Onboard 
9. Other software or services running with RaspAP? 

## Steps to reproduce

Connect to WiFi network with spaces in it's name from Wifi Client web page
## Screenshots

[image]
[image]

## Additional context

wpa_passphrase needs quotation mark ("") when wifi name has spaces in it's name. otherwise it will wrongly parse the input provided by raspap

darnokb, 2023-09-04:
You can of course add wifi manually to /etc/wpa_supplicant/wpa_supplicant.conf, but it's the way it should be done.
Moreover, I did that and tried to remove the wifi network with spaces from raspap-webgui, but I encourtered the same error: 

WPA passphrase must be between 8 and 63 characters

Seems that in configure_client.php quotation marks should be added to exec(); functions where wpa_passphrase is called.

billz, 2023-09-05:
[image]

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
network={
	ssid="POCO F3"
	#psk="test hotspot"
	psk=7ab563ed72732029b8203bb712939d757d5802c55c68e46c11f3f9031b70a27c
}
```

closed 2023-12-12 by github-actions[bot]