RaspAP/raspap-webgui #1397: Get erro "WPA passphrase must be between 8 and 63 characters" even when it is.

dratner, 2023-09-19:
## Checklist

- [ ] This is a bug report
- [ ] I observed this bug on a clean install of the OS
- [ ] I have followed the project prerequisites
- [ ] I have searched this repository for existing issues
- [ ] I checked the FAQ and official documentation before creating this issue
- [ ] I have read and understand the issue reporting guidelines 

## Bug description

When I am trying to connect to a wireless network, I get the error "WPA passphrase must be between 8 and 63 characters" even when the passphrase length is indeed within that range and is working fine on other devices.

## Your environment

1. Operating System: RPi OS 32-bit
2. Hardware and version: Pi 4
3. RaspAP version: RaspAP v2.9.3
4. Clean install of a compatible operating system? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes
7. Simultaneous AP and managed mode? ?
8. Onboard wireless chipset or external adapter? Tried with both wireless chipset and two external adapters
9. Other software or services running with RaspAP? None

## Steps to reproduce

1. Go to Wifi client on Global Nav
2. Find new network under Nearby
3. Enter passphrase
4. Click Add
5. See error (but not for all networks.)

## Screenshots

Although the passphrase is masked in this screenshot, you can see the passphrase is 12 characters, yet I get the error that it's not between 8 and 63. I've seen this on multiple networks, but not all. I've successfully added a few and haven't yet tracked down the difference.

[image]
[image]

## Additional context

billz, 2023-09-19:
Relevant FAQ https://docs.raspap.com/faq/#passphrase

hjai, 2023-09-20:
@dratner I had this same issue. What worked for me is that I had to delete all of the "Known" networks first, then I was able to add the network that gave the error before.

billz, 2023-09-20:
@hjai that's helpful. `knownWifiStations()` reads the contents of your `/etc/wpa_supplicant/wpa_supplicant.conf` file. 

https://github.com/RaspAP/raspap-webgui/blob/10637efee3b5a21c949c4f9bdf61b558d4a7b36e/includes/wifi_functions.php#L8-L11

If this file contains invalid entries, as often happens when manually edited, it will cause the above validation to return the error.

hjai, 2023-09-20:
@billz In my case, the Known network was valid because it was my home wifi network that I had used to initially set up my RaspAP.

dratner, 2023-09-21:
I'll give the gist a try. In the cases where I've had this issue there has been punctuation. The example above was hôtel wifi and I believe the pass phrase was TimeNY@2020 or something similar.

dratner, 2023-09-22:
Ok ... I got executed the gist with the wifi ssid and passphrase that was erroring in the UI. Here is the output: 

network={
	ssid="TIMENY"
	#psk="TimeNY@2020!"
	psk=3351005f802b339ca42ad2e2c27afc6d11aba0bbca18d95034c8c0f8b0576607
}

dratner, 2023-09-22:
PS. I didn't manually update my wpa_supplicant.conf file and it all looks clean, but I'll try reseting it next time I see this error.

dratner, 2023-09-22:
I just tried manually running my wpa_supplicant.conf through the knownWifiStations() function and got what a valid array back. Here is it with passphrases redacted:

$networks = array(
    'CaLeone' => array(
        'visible' => false,
        'configured' => true,
        'connected' => false,
        'index' => 0,
        'ssid' => 'CaLeone',
        'passphrase' => '<redacted long hex>',
        'protocol' => 'WPA',
    ),
    '615Greenleaf' => array(
        'visible' => false,
        'configured' => true,
        'connected' => false,
        'index' => 1,
        'ssid' => '615Greenleaf',
        'passphrase' => '<redacted clear text passphrase>',
        'protocol' => 'WPA-PSK',
    )
);

billz, 2023-09-23:
@dratner I appreciate the thorough testing, Dan. Think the right move here would be to suppress the error message when adding a new WiFi station, in the event knownWifiStations() returns a network with a passphrase that doesn't conform to the character limits. We could instead add an informational message to this effect. Thanks for raising this.

PR #1411 (2023-10-09): Correctly handle 64 hex digit passkey 
A network defined in wpa_supplicant.conf may contain a passphrase, passkey or both. In cases where a known network exists with a valid 64-character passkey, but no passphrase, RaspAP will incorrectly throw a 'WPA passphrase must be between 8 and 63 characters' error.

Resolves #1397

```
# psk: WPA preshared key; 256-bit pre-shared key
# The key used in WPA-PSK mode can be entered either as 64 hex-digits, i.e.,
# 32 bytes or as an ASCII passphrase (in which case, the real PSK will be
# generated using the passphrase and SSID). ASCII passphrase must be bet [... cut]
closed 2023-10-09 by billz