RaspAP/raspap-webgui #1413: Hard block if OpenVPN / Wireguard go offline (ie authentication or error)

frankoz95967943, 2023-10-09:
## Is your feature request related to a problem? 
Love this software.
I noticed the other day RASPAP came up, but openvpn did not.
RASPAP allowed connections and worked even tho openvpn had an error (the password had changed)

## Describe the solution you'd like
Admin option of "Block connections if VPN is in error/disconnected status"
and an autoretry every couple minutes (admin config)
Show alert on main page that VPN is having problems
Send email that vpn is having problems

## Describe alternatives you've considered
Shell script to stop raspap service if OpenVPN had an error on connect / not connected

billz, 2023-10-17:
Something along these lines? https://docs.raspap.com/wireguard/#kill-switch

fabio-stein, 2025-01-13:
Still needing this for OpenVpn connections! I just tested and as soon as the ovpn client stops, it reveals the IP

frankozland, 2025-01-13:
Here's what i ended up with
```
# Flush all counters too #
sudo iptables -Z 
# Flush and delete all nat and  mangle #
sudo iptables -t nat -F
sudo iptables -t nat -X
sudo iptables -t mangle -F
sudo iptables -t mangle -X
sudo iptables -t raw -F
sudo iptables -t raw -X

sudo iptables -A POSTROUTING -j MASQUERADE
#Allow local 192 network
sudo iptables -A POSTROUTING -s 192.168.50.0/24 ! -d 192.168.50.0/24 -j MASQUERADE

#Default allow
sudo iptables -P INPUT ACCEPT
sudo iptables -P FORWARD ACCEPT
sudo iptables -P OUTPUT ACCEPT
sudo iptables -A INPUT -i lo -j ACCEPT
sudo iptables -A INPUT -s 192.168.0.0/16 -j ACCEPT

#Do not allow tun probe
sudo iptables -A FORWARD -i tun+ -o +tun+ -j DROP
#do not allow local net fowrward to tunnel
sudo iptables -A FORWARD -s 192.168.0.0/16 -i tun+ -j DROP

#allow anything that is forwarded, but must go thru OpenVPN tunnel
sudo iptables -A FORWARD -i tun+ -o wlan+ -j ACCEPT
sudo iptables -A FORWARD -i wlan+ -o tun+ -j ACCEPT
sudo iptables -A FORWARD -i eth+ -o tun+ -j ACCEPT
sudo iptables -A FORWARD -i tun+ -o eth+ -j ACCEPT
sudo iptables -A OUTPUT -j ACCEPT
#drop anything that is forwarded that doesnt match any of these rules
sudo iptables -P FORWARD  DROP
#persist settings
sudo netfilter-persistent save
```

dittert, 2025-01-28:
@frankozland how do you run these commands? It does not look like something that is executed whenever the `tun0` interface goes down but something that sets up more general iptables rules. Did you add a new service for that?

I guess there should be a generic way to get interface addresses?