RaspAP/raspap-webgui #1414: Custom VPN set up

frankoz95967943, 2023-10-09:
## Is your feature request related to a problem? 
There are a lot of new VPN products coming to market
ProtonVPN is now available
Wireguard
Lightway
etc

## Describe the solution you'd like
Plug in CLI config to use *ANY* of them
For example, for basic functions like connect, status, disconnect, create a cli config page
Path to executable:
executable name:
command to connect:
command to disconnect:
command to turn on logs:
command to check status:
Frequency to check status: ___ seconds
custom command if disconnected:
custom command when connect:
block traffic if connection lost/error:
email if disconnect: y/n
config file location:
custom options:
install url download:

## Describe alternatives you've considered
This may simplify administration of RASPAP - offer user options to pick a known type (wireguard, openvpn, etc) and prompt user for normal parms, but this would allow new CLI connect for new vpn software.

## Additional context
Add any other context or screenshots about the feature request here.

billz, 2023-10-10:
@frankoz95967943 great suggestion. I reached a similar conclusion while looking at the NordVPN CLI. eg.:

```
Usage: nordvpn [global options] command [command options] [arguments...]

Commands:
     account        Shows account information
     cities         Shows a list of cities where servers are available
     connect, c     Connects you to VPN
     countries      Shows a list of countries where servers are available
     disconnect, d  Disconnects you from VPN
     groups         Shows a list of available server groups
     login          Logs you in
     logout         Logs you out
  ...
```

There is probably enough commonality among popular CLIs to support a subset of the most often used commands.
I see this likely happening at install-time, given that [raspap.sudoers](https://github.com/RaspAP/raspap-webgui/blob/master/installers/raspap.sudoers) would need to be updated with the executable path, eg.:

```
www-data ALL=(ALL) NOPASSWD:/usr/bin/nordvpn *
```

Supported VPN providers and CLIs could be defined in a .json file and read at install time, then used to populate a corresponding web UI at runtime. This also lends itself well to future updates and PRs for new providers.

frankoz95967943, 2023-10-11:
yip very much agree

I'd like to some day be a contributor to this project - im rusty, its been years since i did any meaningful app work - i did see style guide - id love to see if there is a set up for dev env so i can add some to this very cool project

PR #1416 (2023-10-27): Custom VPN provider support
Work in progress. Resolves #1414
closed 2023-10-27 by billz