RaspAP/raspap-webgui #1420: 802.11ac

Ekiben542, 2023-10-16:
## Is your feature request related to a problem? 
802.11ac does not work if country_code is JP.

## Describe the solution you'd like
I would like to be able to use 802.11ac even if the country_code is JP.

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
802.11ac is allowed in Japan.
(https://wifiamateur.blogspot.com/2013/04/80211ac-device-deployment-in-japan.html)Also, this uses Google Translate, so there may be some mistakes, sorry. .

billz, 2023-10-17:
Follow [this FAQ](https://docs.raspap.com/faq/#wificountries) and report your Linux kernel version with `uname -a`.

Ekiben542, 2023-10-17:
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

Ekiben542, 2023-10-17:
The above output is from running uname -a.

billz, 2023-10-18:
I am asking you to read and follow the FAQs on AC support. If you are able to start an 802.11ac AP with 'JP' we can look at adding it.

Ekiben542, 2023-10-18:
Where is the configuration directory?

godfuture, 2023-11-04:
Same for DE....our most famous model (https://avm.de/produkte/fritzbox/fritzbox-7590/technische-daten/) has AC included (as many many others..)

billz, 2023-11-04:
You are both absolutely correct. RaspAP has relied on its own [internal database](https://github.com/RaspAP/raspap-webgui/blob/master/config/wireless.json) to set valid frequencies. This was intended as a short-term fix, but it requires updates with each revision of the kernel's [wireless-regdb](https://wireless.wiki.kernel.org/en/developers/Regulatory/wireless-regdb).

A better approach, I think, would be to parse the output of `iw list` and use this to set allowed frequencies:

```
sudo iw reg set JP
iw list
Wiphy phy1
...
Band 1:
    Frequencies:
	* 2412 MHz [1] (20.0 dBm)
	* 2417 MHz [2] (20.0 dBm)
	* 2422 MHz [3] (20.0 dBm)
	* 2427 MHz [4] (20.0 dBm)
	* 2432 MHz [5] (20.0 dBm)
	...
Band 2:
    Frequencies:
	* 5180 MHz [36] (18.0 dBm)
	* 5200 MHz [40] (18.0 dBm)
	* 5220 MHz [44] (18.0 dBm)
	* 5240 MHz [48] (18.0 dBm)
```
This should in theory make RaspAP's db obsolete.

I'll prototype a solution along these lines. Thanks for raising this.

closed 2023-11-07 by billz