RaspAP/raspap-webgui #1421: raspapd.service can not start dhcpcd.service due to it is not found

velison1-mbed, 2023-10-17:
Hello,

I am trying to use as an access point my raspberry pi zero w. I started with fresh 32 bit lite raspberry pi image. I imaged by rpi imager and install raspap with quick installer and reboot the pi. When I check the wi-fi in my phone, I see raspap-webgui network and tried to connect it but I can't. My phone can not get an ip address. When I check the services (dnsmasq, hostapd and raspapd) I realized that raspapd.service can not start dhcpcd.service. These are raspapd.service logs

-- Boot 1cfd9105d7fd4a648d7ab3e408a433e8 --
Oct 17 08:54:40 raspberrypi systemd[1]: Starting raspapd.service - RaspAP Service Daemon...
Oct 17 08:54:41 raspberrypi bash[805]: Stopping network services...
Oct 17 08:55:10 raspberrypi bash[805]: Changing RaspAP Daemon --interface to uap0
Oct 17 08:55:11 raspberrypi bash[805]: Disabling systemd-networkd
Oct 17 08:55:21 raspberrypi bash[805]: Starting network services...
Oct 17 08:55:25 raspberrypi systemctl[881]: Failed to start dhcpcd.service: Unit dhcpcd.service not found.
Oct 17 08:55:39 raspberrypi bash[805]: RaspAP service start DONE
Oct 17 08:55:39 raspberrypi systemd[1]: raspapd.service: Deactivated successfully.
Oct 17 08:55:39 raspberrypi systemd[1]: Finished raspapd.service - RaspAP Service Daemon.

After that I tried to install dhcpcd with the command "sudo apt-get install dhcpcd5" even that didn’t help. Because there is no dhcpcd.service. There is dhcpcd.conf but no dhcpcd.service.

How can I fix it? Can you help me?

Ekiben542, 2023-10-17:
What is the output of systemctl status dhcpd?

velison1-mbed, 2023-10-17:
> What is the output of systemctl status dhcpd?

Unit dhcpcd.service not found.

Ekiben542, 2023-10-17:
Hmmm, that's a tough one...
If you have an /etc/systemd/system/dhcpcd.service.d
directory, you will find wait.conf there.
[Service].
ExecStart
ExecStart=/usr/sbin/dhcpcd -w -q
I think it would be possible to do this by writing .
If there is no dhcpcd.service.d directory at all, I think you can create a dhcpcd.service.d directory, create a wait.conf and put the above in it. (At your own risk.))

velison1-mbed, 2023-10-17:
> Hmmm, that's a tough one... If you have an /etc/systemd/system/dhcpcd.service.d directory, you will find wait.conf there. [Service]. ExecStart ExecStart=/usr/sbin/dhcpcd -w -q I think it would be [...]

[image]

should I create a new directory in here named dhcpcd.service.d?

Ekiben542, 2023-10-17:
Yes

Ekiben542, 2023-10-17:
Then, if you create wait.conf there and write what you wrote earlier, it will work (Maybe)