RaspAP/raspap-webgui #1440: 2,4 GHz wifi client not connecting, 5 GHz works

godfuture, 2023-11-02:
## Checklist

- [x] This is a bug report
- [x] I observed this bug on a clean install of a supported OS
- [x] I have followed the project prerequisites
- [x] I have searched this repository for existing issues
- [x] I checked the FAQ and official documentation before creating this issue
- [x] I have read and understand the issue reporting guidelines
- [x] I have generated a RaspAP debug log and performed a self-diagnosis

## Bug description

Even the 2,4 GHz wifi from home router was added with correct pass, it does not connect but only show checked status icon and list it under "known". Using a 5 GHz network, the client connects just fine.

## Your environment
1. Operating System: RPi OS 64-bit Lite 
2. Hardware and version: Raspberry Pi 3 B+
3. RaspAP version: RaspAP v2.9.9
4. Clean install of a compatible operating system? Yes
5. RaspAP Quick Install or Manual setup? Quick Install
6. Using default configuration? Yes
8. Onboard wireless chipset or external adapter? Onboard
9. Other software or services running with RaspAP? No (Quick Install incl. OpenVPN, OpenVPN not configured yet)

## Steps to reproduce

Install RPi OS x64
Install RaspAP
Connect ethernet cable
Navigate to WLAN Client
Add your 2,4 GHz network and password

## Screenshots

[image]

## Additional context

billz, 2023-11-02:
Unable to reproduce with info provided.

[image]

You indicated that you generated a RaspAP debug log. What did you find in your self-diagnosis?
Did you try deleting the known network and re-adding it?

godfuture, 2023-11-02:
> Unable to reproduce with info provided.
> [image]
> 
> You indicated that you generated a RaspAP debug log. What did you find in your self-diagnosis? Did you try deleting the known network and re-adding it?

Not really. Only what I already know: "wpa_state=INACTIVE". I wondered wyh I have "ip_address=10.3.141.1" in wpa supplicant, because the wifi client should get the IP of DHCP of my home router.

closed 2023-11-02 by billz

godfuture, 2023-11-02:
Why have you closed it? I followed all the documentation and 2,4 GHz wifi client does not connect.

godfuture, 2023-11-02:
Why so arrogant?

billz, 2023-11-02:
If you cannot/will not provide useful steps to repo, there is nothing to act on.