RaspAP/raspap-webgui #1453: Country code cannot be specified.

Ekiben542, 2023-11-10:
Country code cannot be specified. (↓It looks like this)
 https://github.com/RaspAP/raspap-webgui/assets/71870048/e71b022d-1d85-4386-a3b0-e204b3419aff

## ====Enviroment====
- Debian 12 Bookworm (Raspberry Pi OS 64bit Lite)
- Raspberry Pi 4 Model B
- RaspAP Version 3.0.0 beta

## ====Reproduction Method====
- It occurs when hostapd is restarted after changing any of the passwords, etc.

## ====Additional Context====
- It is possible to do this by manually adding it to hostapd.conf, but it is tedious to manually add the country code every time after changing the configuration.

billz, 2023-11-10:
From the shell try executing `isoquery`.

Ekiben542, 2023-11-10:
-bash: isoquery: command not found

billz, 2023-11-10:
This dependency is handled by the installer. If you pulled the 3.0-beta branch manually, the package wasn't installed.

Add it with: `sudo apt install isoquery`

Ekiben542, 2023-11-10:
Thank you very much. It worked. And the txpower adjustment doesn't work either.

billz, 2023-11-10:
Txpower depends on your wireless reg domain. Most countries limit 5 GHz to 13-18 dBm.

billz, 2023-11-10:
Closing as resolved, thanks

closed 2023-11-10 by billz

Ekiben542, 2023-11-12:
When using the 5Ghz band in Japan, the limit is supposed to be 10dbm, but doesn't this violate the Radio Law?

billz, 2023-11-13:
According to the Linux kernel:

```
iw reg get
global
country JP: DFS-JP

iw phy#1 info
...
* 5180 MHz [36] (18.0 dBm)
* 5200 MHz [40] (18.0 dBm)
* 5220 MHz [44] (18.0 dBm)
* 5240 MHz [48] (18.0 dBm)
```

Ekiben542, 2023-11-13:
but,
https://www2.elecom.co.jp/network/wireless-lan/column/wifi_column/vol08/
of
"About the Radio Law"
In the section of
"Japan's Radio Law sets limits on the output of "wireless stations that do not require a license," such as wireless LAN routers and transceivers. In 2010, the revised Radio Law was promulgated, and this output limit was raised from the previous 10mW to 1000mW. However, this upper limit does not currently apply to wireless LAN routers, and even wireless LAN routers that claim to have "high power" do not exceed the 10mW upper limit if they are officially sold in Japan. must.
This is why you don't often see radio wave output listed in the specifications or catalog specifications of wireless LAN routers, as each manufacturer sets it just within the legal upper limit, so there is no differentiation in terms of radio wave output. Because it cannot be measured."
It says (must be 10dbm or less).
Therefore, I would like to change it, but what should I do? (I personally thought about changing it to 10dbm, but it just says "Operation not permitted" and I can't do it.)

billz, 2023-11-13:
Setting txpower isn't supported by all adapters. It works for the RPi's onboard wireless.

RaspAP executes this:
```
sudo iw dev wlan0 set txpower fixed 10
```

If you're concerned about power limits, use an adapter that supports this.

Ekiben542, 2023-11-13:
When I executed it, I checked with iw wlan1 and found that txpower had not changed (I think the change probably corresponds...)

billz, 2023-11-13:
Not sure what to tell you. try `iw help`. good luck

Ekiben542, 2023-11-14:
sudo iw dev wlan1 set txpower fixed 1000
It was possible by using . thank you

imakiro, 2024-05-24:
Hi, just wanted to inform that I had the same issue, fixed with the installation of "isoquery" AND "iso-codes" on debian bookworm