RaspAP/raspap-webgui #1456: bugs with hotspot and wireguard

ritchieau1, 2023-11-12:
## Checklist

- [ x] This is a bug report
- [x ] I observed this bug on a clean install of a supported OS
- [x ] I have followed the project prerequisites
- [ x] I have searched this repository for existing issues
- [ ] I checked the FAQ and official documentation before creating this issue
- [x ] I have read and understand the issue reporting guidelines
- [ ] I have generated a RaspAP debug log and performed a self-diagnosis

## Bug description

## Your environment
1. Operating System: **ENTER HERE** 
2. Hardware and version: 
3. RaspAP version: 
4. Clean install of a compatible operating system? 
5. RaspAP Quick Install or Manual setup? 
6. Using default configuration? 
8. Onboard wireless chipset or external adapter? 
9. Other software or services running with RaspAP? 

## Steps to reproduce

## Screenshots

## Additional context

ritchieau1, 2023-11-12:
> 1. **ENTER HERE**

Raspberry Pi OS latest version

billz, 2023-11-12:
@ritchieau1 I've updated our issue template with an issue form, which is much easier to use.
I invite you to re-submit your bug report using the new form. Any feedback on the new bug reporting form is appreciated.
Thanks!

closed 2023-11-12 by billz