RaspAP/raspap-webgui #1460: [Bug]: RaspAP web-gui not detecting WiFi stations, but wpa_cli in the shell works

athshean, 2023-11-16:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

External adapter

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.0 (Latest)

### Other software or services running with RaspAP?

No other software

### Bug description

I am using a Ralink Technology, Corp. MT7601U Wireless Adapter usb wifi dongle. I have checked that this is compatible with the Pi using the kernel drivers.

I have completed a fresh setup of the Pi and RaspAP. I use the on-board wifi adapter in AP mode (as the USB adapter does not support AP mode). I then try to use the USB adapter in client mode to connect to a wifi network.

The expected behaviour is that the RasPi GUI should list available wifi networks.

However, I see that Wifi client does not detect any wifi stations. Re-initialising wpa_supplicant does not help either.

I have [looked up the FAQ](https://docs.raspap.com/faq/#wifi-scanning-doesnt-work-or-i-get-the-error-cannot-execute-wpa_cli-reconfigure) as suggested in a previous post, but it seems to me that `wpa_cli` seems to be working from the shell. It is only in the web-gui that it seems to be failing.

### Steps to reproduce

- [ ] Complete fresh install of RaspAP
- [ ] Configure in-built wifi adapter to AP mode
- [ ] Attempt to use USB wifi adapter in client mode to scan for networks

### Relevant log output

```shell
888888ba                              .d888888   888888ba
   88     8b                            d8     88   88     8b
  a88aaaa8P' .d8888b. .d8888b. 88d888b. 88aaaaa88a a88aaaa8P
   88    8b. 88    88 Y8ooooo. 88    88 88     88   88
   88     88 88.  .88       88 88.  .88 88     88   88
   dP     dP  88888P8  88888P  88Y888P  88     88   dP
                               88
                               dP     Debug Log Generator 1.0

This process collects debug and troubleshooting information about your RaspAP installation.
It is intended to assist users with a self-diagnosis of their installations, as well as
provide useful information as a starting point for others to assist with troubleshooting.
[... 304 lines cut ...]
             ├─ 802 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
             ├─ 827 /usr/bin/php-cgi
             ├─ 830 /usr/bin/php-cgi
             ├─ 831 /usr/bin/php-cgi
             ├─ 832 /usr/bin/php-cgi
             ├─ 833 /usr/bin/php-cgi
             ├─1115 sh -c "sudo /etc/raspap/system/debuglog.sh -i /var/www/html"
             ├─1116 sudo /etc/raspap/system/debuglog.sh -i /var/www/html
             ├─1117 /bin/bash /etc/raspap/system/debuglog.sh -i /var/www/html
             ├─1244 /bin/bash /etc/raspap/system/debuglog.sh -i /var/www/html
             └─1245 systemctl status lighttpd

Nov 14 15:32:46 magpie sudo[1026]: pam_unix(sudo:auth): conversation failed
Nov 14 15:32:46 magpie sudo[1026]: pam_unix(sudo:auth): auth could not identify password for [www-data]
Nov 14 15:32:46 magpie sudo[1026]: www-data : command not allowed ; PWD=/var/www/html/ajax/networking ; USER=root ; COMMAND=/usr/sbin/wpa_cli -i wlx90de807ac0a0 scan
Nov 14 15:32:49 magpie lighttpd[1028]: sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper
Nov 14 15:32:49 magpie lighttpd[1028]: sudo: a password is required
Nov 14 15:32:49 magpie sudo[1028]: pam_unix(sudo:auth): conversation failed
Nov 14 15:32:49 magpie sudo[1028]: pam_unix(sudo:auth): auth could not identify password for [www-data]
Nov 14 15:32:49 magpie sudo[1028]: www-data : command not allowed ; PWD=/var/www/html/ajax/networking ; USER=root ; COMMAND=/usr/sbin/wpa_cli -i wlx90de807ac0a0 scan_results
Nov 14 15:32:54 magpie sudo[1116]: www-data : PWD=/var/www/html/ajax/system ; USER=root ; COMMAND=/etc/raspap/system/debuglog.sh -i /var/www/html
Nov 14 15:32:54 magpie sudo[1116]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)

RaspAP debug log generation complete.
```

billz, 2023-11-17:
This project generally cannot support issues related to [external hardware](https://docs.raspap.com/issues/#external-hardware). However, you've provided a complete bug report so we can try to determine if this is an application issue.

Do these commands output a list of SSIDs?

```
sudo wpa_cli -i wlx90de807ac0a0 scan
sudo wpa_cli -i wlx90de807ac0a0 scan_results
```

athshean, 2023-11-19:
> This project generally cannot support issues related to [external hardware](https://docs.raspap.com/issues/#external-hardware). However, you've provided a complete bug report so we can try to [...]

Sorry, I should have been clearer: yes, these commands output a list of SSIDs. That's what I meant by '`wpa_cli` in the shell works' in my post above.

aolson-lg, 2023-11-22:
I have installed RaspAP app on my Raspberry Pi 4 running Debian and found similar problems when I used the Panda PAU0B or PAU0D which is plug and play on Debian Bookworm and also confirmed by Panda technical support. The built-in wireless LAN card on Raspberry Pi 4 is called 'wlan0' and Panda is called 'wlan1'.

I followed the instructions to set Raspberry Pi up as a WiFi repeater (https://docs.raspap.com/repeater/). 
I encountered two issues when I followed 'Connect device to WiFi') with Panda PAU0B or PAU0D plugged into a USB port on the Raspberry Pi:
1) When I go back and forward between web pages, sometimes there is no wireless networks showing up in the WiFi Client page. I need to restart my Raspberry Pi to see the wireless networks again.
2) When I see the available networks in the WiFi Client page, I can add my network but I can't connect to it.

If I manually setup Panda adapter to connect to my network automatically, all I need to do the "Configure routing" section to raise the metric value for wlan0, restart the hotspot. Then I can connect my iPad to my Raspberry Pi and surf the Internet. 
It also works after reboot. :)

From the terminal:
ao@raspberrypi:~ $ hostnamectl
 Static hostname: raspberrypi
       Icon name: computer
      Machine ID: 3cffe4d039694b4e851ce42c0c2a2d89
         Boot ID: c8f56e7793904071943c9d3b4a36dedd
Operating System: Debian GNU/Linux 12 (bookworm)
          Kernel: Linux 6.1.0-rpi6-rpi-v8
    Architecture: arm64

billz, 2023-11-22:
@athshean enabled predictable interface names, attached an external adapter, scanned and connected to a STA via the UI.
logs below:

```
WPA Supplicant
==============
Selected interface 'wlx44a56e63c356'
bssid=f4:ca:e5:a6:be:04
freq=5260
ssid=Freebox-Router
id=0
mode=station
wifi_generation=5
pairwise_cipher=CCMP
group_cipher=CCMP
key_mgmt=WPA2-PSK
wpa_state=COMPLETED
ip_address=192.168.1.34
p2p_device_address=ba:27:eb:01:d7:6a
address=44:a5:6e:63:c3:56
uuid=a13123b3-40df-5de8-9466-f19a68c6eacf
ieee80211ac=1
```

```
USB Devices
===========
Bus 001 Device 004: ID 0846:9053 NetGear, Inc. A6210
```

[image]

It's possible your wpa_supplicant has entered a failed state. wpa_cli has terminate and reassociate commands. If they work for you in the shell, they might be useful to implement in the UI.

closed 2023-12-04 by billz