RaspAP/raspap-webgui #1463: [Bug]:

Mavyre, 2023-11-22:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bullseye

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.0 (Latest)

### Other software or services running with RaspAP?

No other software

### Bug description

It seems that the Bridge AP is having some issue to get set up. When I click "Restart Hotspot", all connection to the Raspberry Pi is lost (can't even SSH to it anymore).
Wifi Hotspot is broadcasted but as ethernet is not responsive anymore, no address is offered to the clients trying to connect to it.
I've ruled out a DHCP issue as the DHCP is offering IP Address to the Raspberry Pi, and the Pi is detected on the network.

I couldn't generate a debug log as I can't acces neither RaspAP Web interface, nor can connect to the Pi through SSH. I'm only able to access the SD Card by mounting it on a computer.

I've search for an answer everywhere both in documentation and FAQ.

### Steps to reproduce

1. Install a clean Raspberry Pi Bullseye 64 bits Lite on SD Card
2. Follow the Quick start guide
3. Install without VPN Server/Client and without Adblocker
4. Go to hotspot > Advanced > Turn on AP Mode
5. Restart hotspot
6. Raspberry Pi network becomes unresponsive

billz, 2023-11-22:
Did you check your router per the docs?

> If you are unable to connect clients to your bridged AP, start by following the recommendations in [this FAQ](https://docs.raspap.com/faq#bridged). Client connectivity issues in bridged AP mode are most often the result of your router, not RaspAP. Check your router's web interface and DHCP settings _before_ reporting a bug.

https://docs.raspap.com/bridged/#troubleshooting

Mavyre, 2023-11-22:
I sure did check all that plus the FAQ about bridged mode instability. I even tried on two different networks to rule out a network config issue.

I've checked, and I'm sure that the DHCP in both networks are really ready to send DHCP leases (other bridged devices, not RaspAPs are working). I can't check the debug log as I don't have a network access to the Pi after switching to Bridge mode... But I can extract logs from the SD Card. Do you know where I can find such a debug log?

Thanks!

billz, 2023-11-22:
You can select Bridged AP, save settings, then generate a debug log before restarting the hotspot.

billz, 2023-11-27:
Started a bridged AP from the UI and connected several clients. Logs below.
RaspAP's debug log generator is incredibly useful  😉 
If possible, verify your upstream router's DHCP settings.

```
System Info
===========
Hardware: Raspberry Pi 3 Model B Rev 1.2
Detected OS: Debian GNU/Linux 12 (bookworm) 64-bit
Kernel: Linux raspberrypi 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux
System Uptime: 0 days, 0 hours, 5 minutes
Memory Usage: 24.8899%

Dhcpcd Contents
===============
# RaspAP default configuration
hostname
clientid
persistent
option rapid_commit
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
option ntp_servers
require dhcp_server_identifier
slaac private
nohook lookup-hostname

# RaspAP br0 configuration
denyinterfaces eth0 wlan0
interface br0

Interfaces
==========
4: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 9e:56:29:c2:02:d4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.18/24 brd 192.168.1.255 scope global dynamic noprefixroute br0
       valid_lft 43008sec preferred_lft 37608sec
    inet6 2a01:e0a:d60:c430:98b9:823d:2bea:97cd/64 scope global dynamic mngtmpaddr noprefixroute
       valid_lft 86202sec preferred_lft 86202sec
    inet6 fe80::442b:9ecc:58cf:c954/64 scope link
       valid_lft forever preferred_lft forever

Routing Table
=============
default via 192.168.1.254 dev br0 proto dhcp src 192.168.1.18 metric 1004
192.168.1.0/24 dev br0 proto dhcp scope link src 192.168.1.18 metric 1004

dhcpcd status:
● dhcpcd.service - DHCP Client Daemon on all interfaces
     Loaded: loaded (/lib/systemd/system/dhcpcd.service; disabled; preset: enabled)
     Active: active (running) since Mon 2023-11-27 09:37:15 GMT; 3min 42s ago
       Docs: man:dhcpcd(8)
    Process: 1059 ExecStart=/usr/sbin/dhcpcd -q -b (code=exited, status=0/SUCCESS)
   Main PID: 1064 (dhcpcd)
      Tasks: 5 (limit: 765)
        CPU: 742ms
     CGroup: /system.slice/dhcpcd.service
             ├─1064 "dhcpcd: [manager] [ip4] [ip6]"
             ├─1065 "dhcpcd: [privileged proxy]"
             ├─1066 "dhcpcd: [network proxy]"
             ├─1067 "dhcpcd: [control proxy]"
             └─1216 "dhcpcd: [BPF ARP] br0 192.168.1.18"

Nov 27 09:37:16 raspberrypi dhcpcd[1065]: br0: soliciting a DHCP lease
Nov 27 09:37:16 raspberrypi dhcpcd[1065]: br0: offered 192.168.1.18 from 192.168.1.254
Nov 27 09:37:17 raspberrypi dhcpcd[1065]: br0: Router Advertisement from fe80::224:d4ff:fea0:52e6
Nov 27 09:37:17 raspberrypi dhcpcd[1065]: br0: adding address 2a01:e0a:d60:c430:98b9:823d:2bea:97cd/64
Nov 27 09:37:17 raspberrypi dhcpcd[1065]: br0: adding route to 2a01:e0a:d60:c430::/64
Nov 27 09:37:17 raspberrypi dhcpcd[1065]: br0: adding default route via fe80::224:d4ff:fea0:52e6
Nov 27 09:37:18 raspberrypi dhcpcd[1065]: br0: probing address 192.168.1.18/24
Nov 27 09:37:23 raspberrypi dhcpcd[1065]: br0: leased 192.168.1.18 for 43200 seconds
Nov 27 09:37:23 raspberrypi dhcpcd[1065]: br0: adding route to 192.168.1.0/24
Nov 27 09:37:23 raspberrypi dhcpcd[1065]: br0: adding default route via 192.168.1.254

systemd-networkd status:
● systemd-networkd.service - Network Configuration
     Loaded: loaded (/lib/systemd/system/systemd-networkd.service; enabled; preset: enabled)
     Active: active (running) since Mon 2023-11-27 09:37:05 GMT; 3min 52s ago
TriggeredBy: ● systemd-networkd.socket
       Docs: man:systemd-networkd.service(8)
             man:org.freedesktop.network1(5)
   Main PID: 1022 (systemd-network)
     Status: "Processing requests..."
      Tasks: 1 (limit: 765)
        CPU: 275ms
     CGroup: /system.slice/systemd-networkd.service
             └─1022 /lib/systemd/systemd-networkd

Nov 27 09:37:05 raspberrypi systemd-networkd[1022]: br0: netdev exists, using existing without changing its parameters
Nov 27 09:37:05 raspberrypi systemd-networkd[1022]: eth0: found matching network '/etc/systemd/network/raspap-br0-member-eth0.network', based on potentially unpredictable interface name.
Nov 27 09:37:05 raspberrypi systemd-networkd[1022]: eth0: Configuring with /etc/systemd/network/raspap-br0-member-eth0.network.
Nov 27 09:37:06 raspberrypi systemd-networkd[1022]: eth0: Gained carrier
Nov 27 09:37:06 raspberrypi systemd-networkd[1022]: eth0: found matching network '/etc/systemd/network/raspap-br0-member-eth0.network', based on potentially unpredictable interface name.
Nov 27 09:37:08 raspberrypi systemd-networkd[1022]: eth0: Gained IPv6LL
Nov 27 09:37:10 raspberrypi systemd-networkd[1022]: wlan0: Gained carrier
Nov 27 09:37:15 raspberrypi systemd-networkd[1022]: br0: Link UP
Nov 27 09:37:15 raspberrypi systemd-networkd[1022]: br0: Gained carrier
Nov 27 09:37:17 raspberrypi systemd-networkd[1022]: br0: Gained IPv6LL
```

closed 2024-01-11 by github-actions[bot]