RaspAP/raspap-webgui #1481: DHCP server page

billz, 2023-12-16:
`(mod_fastcgi.c.449) FastCGI-stderr:PHP Fatal error:  Uncaught TypeError: explode(): Argument #2 ($string) must be of type string, array given in /var/www/html/ajax/networking/get_netcfg.php:14`

_Originally posted by @GitHubUser655321 in https://github.com/RaspAP/raspap-webgui/discussions/1480#discussioncomment-7872501_

billz, 2023-12-24:
@GitHubUser655321 I'm not able to reproduce the error you found.
Can you generate a debug log and paste the `Dnsmasq Contents` section?

GitHubUser655321, 2023-12-25:
If I have time in the next few days I'll do another clean install and see if I can reproduce it.
Here's what I got from the debug log;

```
Dnsmasq Contents
================
/etc/dnsmasq.d/090_raspap.conf
/etc/dnsmasq.d/090_wlan0.conf

/etc/dnsmasq.d/090_raspap.conf contents:
# RaspAP default config
log-facility=/var/log/dnsmasq.log
conf-dir=/etc/dnsmasq.d$

/etc/dnsmasq.d/090_wlan0.conf contents:
# RaspAP wlan0 configuration
interface=wlan0
domain-needed
dhcp-range=10.0.0.100,10.0.0.150,255.255.255.0,12h$
```

GitHubUser655321, 2023-12-29:
I did a clean install of 2023-12-05-raspios-bookworm-arm64 and RaspAP. Didn't change any settings.

Then installed AdGuard Home from here - https://static.adtidy.org/adguardhome/release/AdGuardHome_linux_arm64.tar.gz
Set the web interface to port 3000 and the DNS to listen on 5300.

Back in AdGuard I turn on the 'Only ever query DNS servers configured below' setting and add '127.0.0.1:5300' as the upstream DNS server.

Once I click Save Settings the 'query DNS servers' switch turns off and the settings on the Server Settings page go blank;

[image]

[image]

Here's the full Debug Log;

[raspap_debug-5.log](https://github.com/RaspAP/raspap-webgui/files/13790674/raspap_debug-5.log)

And the conf file from the dnsmasq.d directory
(-rw-r--r-- 1 root root  83 Dec 28 21:33 090_raspap.conf);

[090_raspap.txt](https://github.com/RaspAP/raspap-webgui/files/13790684/090_raspap.txt)

Lemme know if there's anything else I can provide to figure this out.

GitHubUser655321, 2024-01-29:
Today I ran into this same issue _**without**_ AdGuard installed.

Just by flipping the "Only ever query DNS servers configured below" switch and adding an upstream DNS Server (127.0.0.1:5353), the settings on the "DHCP Server" pages got wiped out and none of the settings will save after that.

PR #1508 (2024-01-30): Fix uncaught TypeError: explode()
Resolves #1481
closed 2024-01-30 by billz