RaspAP/raspap-webgui #1482: [Bug]: Uninstaller not behaving correctly

fr3nchalp4a, 2023-12-17:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.0 (Latest)

### Other software or services running with RaspAP?

No other software

### Bug description

The installer must be uninstalling whatever it finds. Currently, the uninstaller fails miserably if something is missing, so it will remove things, fail at some point, and we can't remove things past this (like when the installer failed at some point and the user wants to clean everything before trying to reinstall or just not use RaspAP).

The importance of never having a dirty system is that RaspAP is touching at a lot of the system's critical components (especially network interfaces and routing). If the system is dirty, and configuration files are not correctly installed, the system will broke and the user will lose hours to reinstall RPiOS.

I think that :
- At each step of installation, a log of each step of the install should be made.
- If the installer fails at one moment, ask the user if the script should continue and get past this error or should clean raspap.
- If the user wants the installer to continue, the script simply continues.
- It the user wants to clean this mess, the installer will call the uninstaller with the logfile.
- The uninstaller will take the failed step, try to uninstall as much as possible and revert what it can.
- It will then take in reverse order every install step and run its counterpart.

This approach allow us to better debug installs, as everything is written in a machine log, so we can see which functions are used to install RaspAP. This problems can be avoided by also having an up to date Manual installation wiki! (I tried to install manually and some deps/config files are not there, but I dont have time to file a bug for this.)

### Steps to reproduce

1. Follow the Quick Installer guide on RaspAP website.
2. During install, make the script fail (like Ctrl+C in the middle of an apt install, or better when copying .conf files)
3. See the installer script fail miserably and not even propose to revert previous changes to not leave the system in a dirty state.
4. curl and download locally the install script to be able to use options.
5. `./raspap.sh -n` to try to uninstall RaspAP.
6. Watch the uninstaller also fail miserably when a simple file is missing (because it wasn't even installed in the first time!)

### Relevant log output

```shell
RaspAP Uninstall: Removing sudoers permissions
Removing /etc/sudoers.d/090_raspap
Done.
RaspAP Uninstall: Uninstall completed
Check your network configuration before rebooting to ensure access.
RaspAP Uninstall: Configure uninstall of RaspAP
Detected OS: Debian GNU/Linux 12 (bookworm) 64-bit
RaspAP install directory: /etc/raspap
Lighttpd install directory: /var/www/html? [Y/n]: 
Uninstall from lighttpd directory: /var/www/html
Uninstall RaspAP with these values? [Y/n]: 
RaspAP Uninstall: Removing raspapd.service
Done.
RaspAP Uninstall: Restoring networking config to pre-install defaults
Disabling IP forwarding in /etc/sysctl.d/90_raspap.conf
rm: cannot remove '/etc/sysctl.d/90_raspap.conf': No such file or directory
RaspAP Uninstall Error: Unable to remove /etc/sysctl.d/90_raspap.conf
```

fr3nchalp4a, 2023-12-17:
Also, this could not matter if the installer automatically removes signs of already installed components, but it doesn't actually.

billz, 2023-12-17:
The uninstaller typically lags behind the installer, which is updated more frequently. This project has support for a large number of continuously changing Debian distros so the focus is more often on compatibility.

Rather than complain about how it "fails miserably" perhaps you could instead offer to get involved and improve it. The language you use in this issue is not going to win you any favors.

fr3nchalp4a, 2023-12-18:
I'm terribly sorry about what I said in this issue, I apologize being rude. The installer broke things in my network configuration and it took me a bit of time to reinstall and setup things properly so I was angry about it. (yes I know that I should not install Raspap with other services running in parallel but I can't do otherwise). But I can get an eye on it, since it did failed on some Raspi OS specific things (couldn't copy some config files, which Tailscale, Caddy and Docker did not touch). I close this issue and will make a more complete one.

closed 2023-12-18 by fr3nchalp4a