RaspAP/raspap-webgui #1498: FEATURE REQUEST: Rest API for AP management

NL-TCH, 2024-01-15:
## Is your feature request related to a problem? 
No, but with the RestAPI a whole new world of automation opens, like HA integration with automation based on connected clients, etc...
## Describe the solution you'd like
A Rest API with functionality like:
- get connected clients amount
- list active clients (with details)
- Shutdown AP/ Start AP
- Get VPN status/ details
- Get Adblock status/ details

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.
None
## Additional context
Add any other context or screenshots about the feature request here.

NL-TCH, 2024-01-15:
this is quite a big feature, so idk if this is anything "we" should want

billz, 2024-01-15:
Agreed that the level of effort is somewhat non-trivial. However it's a frequently requested feature #1372 #602 #393 (probably more examples) and would indeed make the project for useful for IoT / HA applications.

I've been gradually moving the project toward more PHP class definitions (see contents of `/src/RaspAP`) so it would make sense for an API to live here. Probably best to start with a few simple endpoints, eg., service status. We could also leverage an existing RESTful API framework, though my preference would be for something lean and minimal rather than a bloated 'kitchen sink' type framework.

NL-TCH, 2024-01-15:
cool!
Is it an idea, that i will try to create a PoC for an API with python (FastAPI)?
So we have a different script for the API on python than on the already existing webgui (on a different port).
Or do you want to have it per-sé integrated in the php code?

NL-TCH, 2024-01-15:
quick update, i have the system info already implemented, moving on to wifi info:

(testing on local machine)
[image]

the following parts i cannot implement because of having zero knowledge about php:
button in webinterface enables a systemd service (which is already installed, but not running)
field in webinterfaces to let the user input a API key (because the API needs to be exposed to the eth0 interface to be of any 

i'll continue this weekend

billz, 2024-01-16:
That was quick 😅 Re: PHP or implementing API endpoints, generally, I'm happy to give advice and/or contribute code as needed.

NL-TCH, 2024-01-26:
Hey @billz the API is progressing quite nicely, can i ask to get insider access?
i am implementing the wireguard api, and it would benefit to have API access to view the different configs imo.

what is your opinion on this?

the branch is quite cluttered, but if you're interested, here is the progress: https://github.com/NL-TCH/raspap-docker/tree/API-alpha

billz, 2024-02-06:
@NL-TCH great! I'll carve out some time and give it a round of testing.

billz, 2024-02-06:
Image built successfully. Started the container per usual. The `raspap-api.service` exited with a path error. Executing the following worked:

```
root@raspberrypi:/home/API# uvicorn main:app --host 0.0.0.0 --port 8081
insecure
INFO:     Started server process [1511]
INFO:     Waiting for application startup.
INFO:     Application startup complete.
INFO:     Uvicorn running on http://0.0.0.0:8081 (Press CTRL+C to quit)
INFO:     192.168.0.101:50949 - "GET /docs HTTP/1.1" 200 OK
INFO:     192.168.0.101:50952 - "GET /openapi.json HTTP/1.1" 200 OK
INFO:     192.168.0.101:51188 - "GET /ap HTTP/1.1" 200 OK
```

Opening http://raspberrypi.local:8081/docs loads the interactive FastAPI docs (very nice). 

Tried several endpoints - all the ones I tested worked as expected. eg.:

GET http://raspberrypi.local:8081/ap 
```
{
  "driver": "nl80211",
  "ctrl_interface": "/var/run/hostapd",
  "ctrl_interface_group": "0",
  "auth_algs": "1",
  "wpa_key_mgmt": "WPA-PSK",
  "beacon_int": "100",
  "ssid": "raspi-webgui",
  "channel": "1",
  "hw_mode": "g",
  "ieee80211n": "1 # 802.11n support (Raspberry Pi 3)",
  "wpa_passphrase": "ChangeMe",
  "interface": "/var/run/hostapd",
  "wpa": "2",
  "wpa_pairwise": "CCMP",
  "country_code": "GB",
  "ignore_broadcast_ssid": ""
}
```

Amazing work!

PR #1519 (2024-03-09): RestAPI installer integrated
Hi,

I am creating this PR because: 
- all the GET requests are in place
- i am getting really busy next months
- i am a little burned out on this project :)

Feel free to create more post request functions etc...

Gr. TCH

*this is also tested in docker
/resolves #1498 #1507 #1372 #602 #393
closed 2024-03-09 by billz