RaspAP/raspap-webgui #1502: [Bug]: changing wifi password via config does not work

NL-TCH, 2024-01-20:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.0 (Latest)

### Other software or services running with RaspAP?

Yes (specify below)

### Contact details (optional)

discord: TCH

### Bug description

when using https://gist.github.com/billz/2cc43e96563293d650e313e068d52dfb or manually editing the `/etc/hostapd/hostapd.conf` file the SSID can be changed. but when changing the wifi password, clients cannot connect anymore with the old nor new password. is this a bug?!

### Steps to reproduce

change the `wpa_passphrase` in `/etc/hostapd/hostapd.conf ` and reload the daemon + raspapd services.

**again, changing ssid works, but password not!**

NL-TCH, 2024-01-20:
owh this might have to do something with setting a custom DHCP pool, then removing the docker container.
the WLAN0 settings of the adapter stay, but the container is removed, when creating  a new raspap container, nothing can connect anymore.

maybe we need a cleanup script for on the docker host ....
i'll wait for that before i can continue working on the API, i don't want to reinstall my pi :(

NL-TCH, 2024-01-20:
uncommenting the static config in `/etc/dhcpcd.conf` and reinstalling the docker container gives the interface it's ip's back:
```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.3.141.1  netmask 255.255.255.0  broadcast 10.3.141.255
        inet6 fe80::3a7f:e86a:3329:a5f8  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:41:62:38  txqueuelen 1000  (Ethernet)
        RX packets 13  bytes 1573 (1.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 179  bytes 30160 (29.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

but still clients cannot connect

NL-TCH, 2024-01-20:
even when running the uninstaller on container1, deleting the container, running a new docker container with raspap. containers cannot connect

billz, 2024-01-21:
In a Docker context, I think this can be resolved by adding a volume for config files:

```
- ./raspap-conf/hostapd.conf:/etc/hostapd/hostapd.conf:rw
```

(untested)

closed 2024-01-27 by NL-TCH