RaspAP/raspap-webgui #1507: [NOT a Bug]: create equal access rights for /etc/wireguard as /etc/openvpn

NL-TCH, 2024-01-27:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 5

### RaspAP version

3.0 (Latest)

### Other software or services running with RaspAP?

Yes (specify below)

### Contact details (optional)

discord: TCH

### Bug description

related to #1498 

/etc/wireguard is privileged, so the API cannot read the configs without privilege (which is not ideal).

```
tch@raspberrypi:/etc $ ls -al | grep wire
drwx------   2 root     root      4096 Jan 27 15:32 wireguard
tch@raspberrypi:/etc $ ls -al | grep openvpn
drwxr-xr-x   4 root     root      4096 Jan 20 19:30 openvpn
tch@raspberrypi:/etc $ 
```

my request: make /etc/wireguard the same access profile as /etc/openvpn (in the install script)

### Steps to reproduce

 ls -al | grep wire
ls -al | grep openvpn

NL-TCH, 2024-01-27:
same for ddnsclient config:

```
tch@raspberrypi:/etc $ ls -al |grep ddclient.conf 
-rw-------   1 root     root       199 Jan 27 17:19 ddclient.conf

```

NL-TCH, 2024-01-27:
Thanks for providing insider access btw!

just a question not related to this issue: how do i list the connected clients in bash? i can list DHCP leases, but i have tried anything in the php files without any luck.

the only method i can come up with is:
```mermaid
flowchart TD
    A["cat /var/log/dnsmasq.log | grep DHCPACK"] -->|"where time is less then leasetime"| B(Active clients)
```

but this is not a list of active clients, but a list of active leases...

billz, 2024-01-28:
> Thanks for providing insider access btw!

@NL-TCH you betcha :)

Good question re: active leases. Check out the related discussions in https://github.com/RaspAP/raspap-webgui/issues/538 and https://github.com/RaspAP/raspap-webgui/pull/302

> Instead of using dnsmasq leases, I'm leaning towards using iw station dump to collect MAC addresses, then cross referencing them with wlan0's arp table. Then some more awk/sed/grep magic to get the relevant data.

NL-TCH, 2024-02-05:
/tmp/hostapd.log cannot be accessed by a normal user (and API) as well

billz, 2024-02-06:
> /etc/wireguard is privileged, so the API cannot read the configs without privilege (which is not ideal).

I might be wrong, but modifying wg permissions likely opens some big security holes. Defaults are:

700 for the /etc/wireguard directory
600 for the contents of /etc/wireguard

```
$ ls -ld /etc/wireguard
drwx------ 2 root root 4096 Sep 19  2022 /etc/wireguard
```

/etc/ddclient.conf also contains passwords.

Exposing these to an API is an open invitation to 'security researchers' looking for their first CVE.

NL-TCH, 2024-02-20:
fixed by using "sudo"

closed 2024-02-20 by NL-TCH