RaspAP/raspap-webgui #1514: Invalid default DHCP configuration for wlan0 interface

esackowitz, 2024-02-08:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(32-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.0 (Latest)

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

eric@secureco.com

### Bug description

**After a clean install of the O/S using Raspberry Pi Imager** 

[image]

**then performing all needed updates and following the automated install of RASPAP**

[image]

**Out of the box the default DHCP Server setup on Wlan0 is missing network start and end IP ranges along with lease time and interval.  This is an invalid configuration.**

[image]

**Any attempt to save as is will fail with error as expected**

[image]

  **Any attempt at manually entering in the default ranges starting with 10.3.141.50 to 10.3.141.254 and the 12hr lease interval will allow you to save the settings and will confirm as valid,  however, those settings do not get applied and the screen will revert to the original invalid config.**

### Steps to reproduce

1. Install clean O/S using raspberry Pi imager on SD card (this was done on a MACBook Pro)
2. Boot Raspberry Pi with that image
3. Ethernet cable is plugged direct into switch
4. SSH in and Apply all updates and reboot:
**sudo apt-get update
sudo apt-get full-upgrade
sudo reboot**
[... 252 lines cut ...]
        CPU: 12.005s
     CGroup: /system.slice/lighttpd.service
             ├─  842 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
             ├─  862 /usr/bin/php-cgi
             ├─  863 /usr/bin/php-cgi
             ├─  864 /usr/bin/php-cgi
             ├─  865 /usr/bin/php-cgi
             ├─  866 /usr/bin/php-cgi
             ├─16170 sh -c "sudo /etc/raspap/system/debuglog.sh -i /var/www/html"
             ├─16171 sudo /etc/raspap/system/debuglog.sh -i /var/www/html
             ├─16172 /bin/bash /etc/raspap/system/debuglog.sh -i /var/www/html
             ├─16295 /bin/bash /etc/raspap/system/debuglog.sh -i /var/www/html
             └─16296 systemctl status lighttpd

Feb 07 23:19:17 secureco2f727eda sudo[16037]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Feb 07 23:19:17 secureco2f727eda lighttpd[16038]: /bin/chmod: cannot access '/tmp/hostapd.log': No such file or directory
Feb 07 23:19:17 secureco2f727eda sudo[16037]: pam_unix(sudo:session): session closed for user root
Feb 07 23:19:25 secureco2f727eda sudo[16098]: www-data : PWD=/var/www/html ; USER=root ; COMMAND=/bin/chmod o+r /var/log/dnsmasq.log
Feb 07 23:19:25 secureco2f727eda sudo[16098]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Feb 07 23:19:25 secureco2f727eda sudo[16098]: pam_unix(sudo:session): session closed for user root
Feb 07 23:19:39 secureco2f727eda sudo[16171]: www-data : PWD=/var/www/html/ajax/system ; USER=root ; COMMAND=/etc/raspap/system/debuglog.sh -i /var/www/html
Feb 07 23:19:39 secureco2f727eda sudo[16171]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Feb 07 23:19:39 secureco2f727eda lighttpd[16292]: Unit wg-quick@wg0.service could not be found.
Feb 07 23:19:39 secureco2f727eda lighttpd[16294]: Unit openvpn-client@client.service could not be found.
```

billz, 2024-02-08:
@esackowitz thanks for the very complete and accurate bug report. As a gesture of gratitude, I'd like to offer you Insiders access for your time.

closed 2024-02-08 by billz

esackowitz, 2024-02-10:
> @esackowitz thanks for the very complete and accurate bug report. As a gesture of gratitude, I'd like to offer you Insiders access for your time.

You're very welcome. Thank you for the quick turn.  Happy to join the insiders.

esackowitz, 2024-02-10:
I would be remiss to report that the fix worked in saving the DHCP values, however, DNSMasq wont start. 
[image]
Happy to fill out another issue if you prefer.

billz, 2024-02-10:
Please generate a debug log and perform a self-diagnosis before creating a new issue, thanks.

PR #1515 (2024-02-08): Update parsing of dhcp-range
Resolves #1514