RaspAP/raspap-webgui #1523: Insider repo not found

SDollery, 2024-02-22:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 3 Model B+

### RaspAP version

3.0 (Latest)

### Other software or services running with RaspAP?

Yes (specify below)

### Contact details (optional)

shane@dollery.net.au

### Bug description

I am trying to download the insider version but I'm receiving the following error when running the script in the guide:

curl -sL https://install.raspap.com | bash -s -- --upgrade --insiders --name sdollery  --token [Token Removed]

fatal: repository 'https://github.com/RaspAP/raspap-insiders/' not found

### Steps to reproduce

Run the following upgrade script per guide here: https://docs.raspap.com/insiders/#installing

curl -sL https://install.raspap.com | bash -s -- --upgrade --insiders --name [Name Removed]  --token [Token Removed]

### Screenshots

[image]

billz, 2024-02-22:
When you become a sponsor, GitHub sends you an invite to the private Insiders repo. You must accept this invite before upgrading your install. Your invite still appears as "pending" and waiting for your response:

[image]

Check your various mail folders to be sure you didn't overlook the invite from GitHub.

billz, 2024-02-22:
You can verify your token beforehand by using the method described here:
https://docs.raspap.com/insiders/#authentication

The above link also appears in the installer for Insiders.

SDollery, 2024-02-22:
Hi Bill,

Thanks. There was a notification in GitHub but no email received. Clicked the accept in the message directly in GitHub and that fixed it for me. Now to have a look at the extra features and see if I can get my 4G dongle working. Then some sort of least cost routing. Thanks very much for the help!

Regards,

Shane

image.png (view on web)<https://github.com/RaspAP/raspap-webgui/assets/229399/49528d0b-5937-4c7e-951a-6c7b9a00cc58>

billz, 2024-02-22:
Glad to hear it. Marking this as closed.

closed 2024-02-22 by billz