RaspAP/raspap-webgui #1570: [Bug]: wlan0 has no static IP; dhcpcd fails on boot

MukiEX, 2024-04-26:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi Zero 2 W

### RaspAP version

3.0 (Latest)

### Other software or services running with RaspAP?

No other software

### Bug description

Basically it looks like the dhcpcd service fails on boot and wlan0 has no IPs. In addition, when connecting to the AP SSID, the client device gets no IPv4 address set

### Steps to reproduce

- Fresh install of Raspberry Pi OS 12
- Run the prerequisites for RaspAP install (get apt packages up to date 'n set Wi-Fi Country (US in my case) )
- Fresh curl install of RaspAP
  - Skip VPN/RestAPI portion of installation (forgot to do the latter on the first go-around, ran a 2nd install pass)

ssh into RaspAP device
Run `ip a s wlan0`
No IPs present
Check `systemctl status dhcpcd`

### Additional context

For what it's worth, this install procedure seems to work fine on Raspberry Pi OS 11, and OS 12 has been out for a while, and googlin' only turned up an earlier, similar issue, but no workaround I could find, and it seems that issue has been fixed since.

### Relevant log output

```shell
# Added 100 lines of dmesg and 200 lines of "journalctl -u dhcpcd" output at the bottom, after "RaspAP debug log generation complete."
# Labeled "# dmesg output:" and "# journalctl output:", respectively

   888888ba                              .d888888   888888ba
   88     8b                            d8     88   88     8b
  a88aaaa8P' .d8888b. .d8888b. 88d888b. 88aaaaa88a a88aaaa8P
   88    8b. 88    88 Y8ooooo. 88    88 88     88   88
   88     88 88.  .88       88 88.  .88 88     88   88
   dP     dP  88888P8  88888P  88Y888P  88     88   dP
                               88
[... 537 lines cut ...]
Apr 25 22:40:39 hpe-print dhcpcd[1352]: ps_ctl_recv: read: Success
Apr 25 22:40:48 hpe-print dhcpcd[1352]: ps_sendcmdmsg: Connection refused
Apr 25 22:40:48 hpe-print dhcpcd[1352]: ps_inet_recvra: Connection refused
Apr 25 22:42:09 hpe-print systemd[1]: dhcpcd.service: State 'stop-sigterm' timed out. Killing.
Apr 25 22:42:09 hpe-print systemd[1]: dhcpcd.service: Killing process 1352 (dhcpcd) with signal SIGKILL.
Apr 25 22:42:09 hpe-print systemd[1]: dhcpcd.service: Failed with result 'signal'.
Apr 25 22:42:09 hpe-print systemd[1]: dhcpcd.service: Scheduled restart job, restart counter is at 10.
Apr 25 22:42:09 hpe-print systemd[1]: Stopped dhcpcd.service - DHCP Client Daemon.
Apr 25 22:42:10 hpe-print systemd[1]: Starting dhcpcd.service - DHCP Client Daemon...
Apr 25 22:42:10 hpe-print dhcpcd[1376]: main: control_open: Connection refused
Apr 25 22:42:10 hpe-print dhcpcd[1376]: main: control_open: Connection refused
Apr 25 22:42:10 hpe-print dhcpcd[1376]: dhcpcd-9.4.1 starting
Apr 25 22:42:10 hpe-print dhcpcd[1379]: dev: loaded udev
Apr 25 22:42:10 hpe-print dhcpcd[1379]: dhcp6_openudp: Address already in use
Apr 25 22:42:10 hpe-print dhcpcd[1376]: dhcp6_openudp: Address already in use
Apr 25 22:42:10 hpe-print dhcpcd[1376]: ps_inet_startcb: dhcp6_open: Address already in use
Apr 25 22:42:10 hpe-print dhcpcd[1379]: ps_inet_startcb: dhcp6_open: Address already in use
Apr 25 22:42:10 hpe-print dhcpcd[1379]: DUID 00:01:00:01:2d:bd:be:60:2c:cf:67:1f:2a:98
Apr 25 22:42:10 hpe-print systemd[1]: Started dhcpcd.service - DHCP Client Daemon.
Apr 25 22:42:10 hpe-print dhcpcd[1379]: wlan0: connected to Access Point:
Apr 25 22:42:10 hpe-print dhcpcd[1379]: eth0: IAID c6:87:7a:59
Apr 25 22:42:10 hpe-print dhcpcd[1379]: ps_ctl_listen: read: Success
Apr 25 22:42:10 hpe-print dhcpcd[1379]: ps_ctl_recv: read: Success
Apr 25 22:42:10 hpe-print systemd[1]: dhcpcd.service: Main process exited, code=killed, status=31/SYS
```

billz, 2024-04-26:
Fairly obvious that DHCP is failing to start due to this:

```
dhcpcd[1379]: dhcp6_openudp: Address already in use
```

This indicates that ipv6 is enabled, perhaps done by your HP printer (or whatever) software that you've installed. This is not supported by RaspAP. As this appears to be an unclean OS, try again without your printer software and be sure ipv6 is disabled.

MukiEX, 2024-04-27:
> Fairly obvious that DHCP is failing to start due to this:
>
> ```
> dhcpcd[1379]: dhcp6_openudp: Address already in use
> ```
>
> This indicates that ipv6 is enabled, perhaps done by your HP printer (or [...]

1. Thank you very much for taking a look.
2. I was confused at first, but then I realized what I did. The server's hostname is hpe-print, which I set via the Raspberri Pi USB image writer, but I haven't actually installed any HP software or even plugged any printers into it yet. Eventually I'm going to have it at my office and the hostname was to easily identify it.
3. For what it's worth, I disabled ipv6 via /etc/sysctl.conf and rebooted, but the dhcpcd service still fails:

`-- Boot d189bf84873e4d87b97c13f966e89e4d --
Apr 26 19:48:58 hpe-print systemd[1]: Starting dhcpcd.service - DHCP Client Daemon...
Apr 26 19:48:58 hpe-print dhcpcd[460]: dhcpcd-9.4.1 starting
Apr 26 19:48:58 hpe-print dhcpcd[471]: dev: loaded udev
Apr 26 19:48:58 hpe-print dhcpcd[471]: DUID 00:01:00:01:2d:bd:be:60:2c:cf:67:1f:2a:98
Apr 26 19:48:58 hpe-print systemd[1]: Started dhcpcd.service - DHCP Client Daemon.
Apr 26 19:48:58 hpe-print dhcpcd[471]: eth0: waiting for carrier
Apr 26 19:49:00 hpe-print dhcpcd[471]: wlan0: waiting for carrier
Apr 26 19:49:00 hpe-print dhcpcd[471]: wlan0: carrier acquired
Apr 26 19:49:00 hpe-print dhcpcd[471]: wlan0: IAID 67:1f:2a:98
Apr 26 19:49:00 hpe-print dhcpcd[471]: ps_ctl_listen: read: Success
Apr 26 19:49:00 hpe-print dhcpcd[471]: ps_ctl_recv: read: Success
Apr 26 19:49:00 hpe-print systemd[1]: dhcpcd.service: Main process exited, code=killed, status=31/SYS
Apr 26 19:49:03 hpe-print dhcpcd[471]: ps_sendcmdmsg: Connection refused
Apr 26 19:49:03 hpe-print dhcpcd[471]: ps_inet_recvra: Connection refused`

4. I'll try another install pass with defaults on all of that, the user/hostname mighta futzed things up and I can change it after everything's verified working anyway.
5. Again, appreciate the help. I was pulling my hair on this one.

MukiEX, 2024-04-29:
[2024-04-28_rpi12_raspap_debug.log](https://github.com/RaspAP/raspap-webgui/files/15144859/2024-04-28_rpi12_raspap_debug.log)

Not sure if this will be helpful, but here's as much info as I can provide for the config:

1. Fresh install using Raspberry Pi Imager on Pi Zero 2 W
2. Raspberry Pi OS 12 (Bookworm) Lite (32-Bit) (default settings, no hostname change)
3. Plugged ethernet USB device for DHCP web interface address 'n a logitech keyboard (wireless usb connector), turned on
4. Ran `raspi-config` and set time zone to US/Central, Wi-fi country to US, and keyboard to US (pipe key was different and tripped me up)
5. Ran `apt update` and `apt full-upgrade` (rebooted)
6. Ran `curl -sL https://install.raspap.com | bash`
7. Chose no for VPN, adblock, that one guard option, and restAPI.

For what it's worth, it looks like the three issues are still present:

1. dhcpcd still fails to start
2. wlan0 has no IP
3. ipv6 still listed on eth0

New log attached

billz, 2024-04-29:
Your wlan0 has no ipv4 address. You'll need to resolve this before dhcpcd will start. I cannot regress this issue further until this is done. No hard feelings.

MukiEX, 2024-05-01:
It's all good, no prob. I thought I had it set in the DHCP server section, or at least that it's set by default. Tried re-applying the address, but it doesn't seem to fix it. Knowing my luck it's gonna end up being some weird esoteric bug on the Pi Zero 2 wi-fi chip driver they fix in Pi OS 13 or something.

[image]

In any event, I appreciate the help.

billz, 2024-05-01:
Setting a static IP via the UI will only configure dhcpcd.conf. It will not resolve the fundamental problem with your adapter. 

Your zero may have inadequate voltage for the WLAN adapter. Disconnect your wireless keyboard and any USB hubs. Prepare a clean OS and confirm that wlan0 has an ipv4 address _before installing anything_.

This issue is straying into general troubleshooting territory, which unfortunately I don't have time for. Suggest your join our Discord server and ask the community for help.

stefan-matic, 2024-05-02:
I'm experiencing the same kind of problem with `wlan0` not having an IP address.

I've reimaged a couple of times, tested different chargers, different sd cards, different raspberries, ran the installers with different options, tried Raspbian Bookworm and Bullseye (both Lite) but every time I have the same problem with `wlan0` not having an IP address set.

billz, 2024-05-02:
@stefan-matic please read this https://docs.raspap.com/issues/#guidelines

stefan-matic, 2024-05-02:
### Code of Conduct

-   [x]  I agree to follow this project's Code of Conduct.

### Issue reporting checklist

-   [x]  I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
-   [x]  I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
-   [x]  I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
-   [x]  I have searched this repository for existing issues.
-   [x]  I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
-   [x]  I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS (32-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4B (4GB RAM)

### RaspAP version

[3.1.3](https://github.com/RaspAP/raspap-webgui/releases/tag/3.1.3) (Latest)

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

stefan@matic.ba

### Bug description

`wlan0` has no IP on fresh install.
Cannot connect to the raspi-webgui SSID via Android phones or PCs (Windows, MacOS or Linux)

### Steps to reproduce

-   Fresh install of Raspbian Lite (32-bit) (Bookworm)
-   Run the prerequisites for RaspAP install (get apt packages up to date and set Wi-Fi Country (BA in my case) )
-   Fresh curl install of RaspAP using `curl -sL https://install.raspap.com | bash` (Used all default options and NordVPN)
-   Start Hotspot

ssh into RaspAP device\
Run `ip a s wlan0`\
No IPs present\
Check `systemctl status dhcpcd`

### Screenshots

[image]
[image]
[image]

### Additional context

I'm connecting `eth0` to my main router.
`wlan0` is used for hotspot

- Tried different RPIs
[... 297 lines cut ...]
             ├─ 856 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
             ├─ 876 /usr/bin/php-cgi
             ├─ 877 /usr/bin/php-cgi
             ├─ 878 /usr/bin/php-cgi
             ├─ 879 /usr/bin/php-cgi
             ├─ 880 /usr/bin/php-cgi
             ├─1752 sh -c "sudo /etc/raspap/system/debuglog.sh -i /var/www/html"
             ├─1753 sudo /etc/raspap/system/debuglog.sh -i /var/www/html
             ├─1754 /bin/bash /etc/raspap/system/debuglog.sh -i /var/www/html
             ├─1879 /bin/bash /etc/raspap/system/debuglog.sh -i /var/www/html
             └─1880 systemctl status lighttpd

May 02 23:30:52 fallen-ap sudo[1671]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
May 02 23:30:52 fallen-ap sudo[1671]: pam_unix(sudo:session): session closed for user root
May 02 23:30:52 fallen-ap sudo[1674]: www-data : PWD=/var/www/html/ajax/networking ; USER=root ; COMMAND=/sbin/wpa_cli -i wlan0 scan
May 02 23:30:52 fallen-ap sudo[1674]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
May 02 23:30:52 fallen-ap sudo[1674]: pam_unix(sudo:session): session closed for user root
May 02 23:30:55 fallen-ap sudo[1678]: www-data : PWD=/var/www/html/ajax/networking ; USER=root ; COMMAND=/sbin/wpa_cli -i wlan0 scan_results
May 02 23:30:55 fallen-ap sudo[1678]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
May 02 23:30:55 fallen-ap sudo[1678]: pam_unix(sudo:session): session closed for user root
May 02 23:31:02 fallen-ap sudo[1753]: www-data : PWD=/var/www/html/ajax/system ; USER=root ; COMMAND=/etc/raspap/system/debuglog.sh -i /var/www/html
May 02 23:31:02 fallen-ap sudo[1753]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)

RaspAP debug log generation complete.
```