RaspAP/raspap-webgui #1571: [Bug]: Saving any settings triggers password manager extention to update password for the portal

navado, 2024-04-26:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Debian	Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Other

### RaspAP version

3.0 (Latest)

### Other software or services running with RaspAP?

Yes (specify below)

### Bug description

When saving any settings from web ui, password manager triggered to update password. 

### Steps to reproduce

Open web interface from Firefox with LastPass and try to configure any setting

### Screenshots

[image]

billz, 2024-04-26:
RaspAP uses basic auth for the admin login. You'll need to check if LastPass supports this or if there's an option to suppress this dialog. We can't offer any support for third-party browser plugins.

closed 2024-04-26 by billz

navado, 2024-04-26:
It's surely supported, login works like charm.
Just means that UI is annoying when used with browser with Last Pass installed. 
Is it possible to change authentication method in general? isn't it reasonable feature request?

billz, 2024-04-26:
Ok, one complaint I've heard is that pw managers aren't able to deal with basic auth (pretty sure 1Password doesn't). Re: changing the auth method, with enough time and resources anything is possible 😉 I don't have bandwidth to address this now, but PR's are always welcome.