RaspAP/raspap-webgui #1585: problem with AVM fritzbox wiregurad config [Bug]:

er34zer, 2024-05-12:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

Yes (specify below)

### Contact details (optional)

p.bartelt@gmx.de

### Bug description

every time load the Wireguard configfile from a Fritzbox i get a Bug:

"
An exception occurred
Stack trace:

TypeError: htmlspecialchars(): Argument #1 ($string) must be of type string, array given
 at htmlspecialchars(general.php:96)
 ... 7 more

"

after that no new .conf can be loaded. The only way is > format SD > load RaspiOSlite > install RaspAP via SSH...

### Steps to reproduce

1. Flash Raspberry Pi OS with Raspberry Pi Imager
2. SD -> Raspi 4 and bootup
3. Connect Via Ehernet & SSH (Putty)
4. Run APT Update & Upgrade
5. Set Location in Raspiconfig
6. Run Curl for RaspAP
7. Connect via Ethernet, Firefox, HTTPS 
8. Load the Wireguard Config from the bussines AVM Fritzbox 15km away
9. ERROR

[... 317 lines cut ...]
             ├─  1051 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
             ├─  1107 /usr/bin/php-cgi
             ├─  1116 /usr/bin/php-cgi
             ├─  1117 /usr/bin/php-cgi
             ├─  1118 /usr/bin/php-cgi
             ├─  1119 /usr/bin/php-cgi
             ├─119551 sh -c "sudo /etc/raspap/system/debuglog.sh -i /var/www/html"
             ├─119552 sudo /etc/raspap/system/debuglog.sh -i /var/www/html
             ├─119553 /bin/bash /etc/raspap/system/debuglog.sh -i /var/www/html
             ├─119676 /bin/bash /etc/raspap/system/debuglog.sh -i /var/www/html
             └─119677 systemctl status lighttpd

May 12 15:46:33 Raspap sudo[118805]: pam_unix(sudo:session): session closed for user root
May 12 15:47:02 Raspap lighttpd[118855]: sh: 1: host: not found
May 12 15:47:02 Raspap lighttpd[118865]: sh: 1: host: not found
May 12 15:47:02 Raspap lighttpd[118875]: sh: 1: host: not found
May 12 15:47:14 Raspap lighttpd[118927]: basename: fehlender Operand
May 12 15:47:14 Raspap lighttpd[118927]: „basename --help“ liefert weitere Informationen.
May 12 16:21:11 Raspap lighttpd[119474]: Aufruf: grep [OPTION]… MUSTER [DATEI]…
May 12 16:21:11 Raspap lighttpd[119474]: „grep --help“ liefert weitere Informationen.
May 12 16:21:20 Raspap sudo[119552]: www-data : PWD=/var/www/html/ajax/system ; USER=root ; COMMAND=/etc/raspap/system/debuglog.sh -i /var/www/html
May 12 16:21:20 Raspap sudo[119552]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)

RaspAP debug log generation complete.
```

billz, 2024-05-12:
If your wg config is of a nonstandard format, the parser may fail to handle it. Please share your wg0.conf (with keys removed) similar to this https://github.com/RaspAP/raspap-webgui/discussions/1430#discussioncomment-7373035

Previously fixed with https://github.com/RaspAP/raspap-webgui/pull/1432

er34zer, 2024-05-12:
hi,

here it is, ist direct from the FritzBox and works on a Win11 PC with Wireguard (if it helps)

`[Interface]
PrivateKey = [PrivateKey]
Address = 192.168.178.203/24
DNS = 192.168.178.1
DNS = fritz.box

[Peer]
PublicKey = [PublicKey]
PresharedKey = [PresharedKey]
AllowedIPs = 192.168.178.0/24,0.0.0.0/0
Endpoint = [Endpoint]
PersistentKeepalive = 25`

billz, 2024-05-12:
Thanks. RaspAP expects a single DNS value in the client config, see [this as an example](https://wiki.archlinux.org/title/WireGuard#Client_configuration 
). Multiple values are unusual, hence the error. In the short term, removing the second DNS value should allow you to start the wg service (I'm guessing the FQDN resolves to your router's IPv4 address).

er34zer, 2024-05-12:
Hi again,
thanks 4 support! But its still nor running

ive changed the .conf like the linked example:

`  
[Interface]
PrivateKey = [PrivateKey]
Address = 192.168.178.203/24
DNS = 192.168.178.1

[Peer]
PublicKey = [PublicKey]
PresharedKey = [PresharedKey]
Endpoint = [Endpoint]
AllowedIPs = 192.168.178.0/24,0.0.0.0/0
`

Wireguard Log:

`
Mai 12 20:09:25 Raspap wg-quick[919]: wg-quick: `/etc/wireguard/wg0.conf' does not exist
Mai 12 20:11:39 Raspap wg-quick[1256]: [#] ip link add wg0 type wireguard
Mai 12 20:11:39 Raspap wg-quick[1256]: [#] wg setconf wg0 /dev/fd/63
Mai 12 20:11:39 Raspap wg-quick[1256]: [#] ip -4 address add 192.168.178.203/24 dev wg0
Mai 12 20:11:39 Raspap wg-quick[1256]: [#] ip link set mtu 1420 up dev wg0
Mai 12 20:11:39 Raspap wg-quick[1299]: [#] resolvconf -a tun.wg0 -m 0 -x
Mai 12 20:11:39 Raspap wg-quick[1256]: [#] wg set wg0 fwmark 51820
Mai 12 20:11:39 Raspap wg-quick[1256]: [#] ip -4 route add 0.0.0.0/0 dev wg0 table 51820
Mai 12 20:11:39 Raspap wg-quick[1256]: [#] ip -4 rule add not fwmark 51820 table 51820
Mai 12 20:11:39 Raspap wg-quick[1256]: [#] ip -4 rule add table main suppress_prefixlength 0
Mai 12 20:11:39 Raspap wg-quick[1256]: [#] sysctl -q net.ipv4.conf.all.src_valid_mark=1
Mai 12 20:11:39 Raspap wg-quick[1256]: [#] nft -f /dev/fd/63
Mai 12 20:11:39 Raspap wg-quick[1256]: [#] iptables -A FORWARD -i wlan0 -o wg0 -j ACCEPT; iptables -A FORWARD -i wg0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT; iptables -t nat -A  POSTROUTING -o wg0 -j MASQUERADE
`

billz, 2024-05-12:
The first line in the service output is the clue:
```
wg-quick: `/etc/wireguard/wg0.conf' does not exist
```

You can either rename this file, or edit it on another device and (re)upload it via RaspAP. This will ensure that the file is saved in the correct location.

Using 3 backticks to format code blocks ``` makes output easier to read 😉

er34zer, 2024-05-12:
hi

ok, i just clicked on the Code Button...

next try by renamed it t wg0.conf resolved:

```
Mai 12 23:04:00 Raspap wg-quick[1734]: [#] ip -4 rule delete table 51820
Mai 12 23:04:00 Raspap wg-quick[1734]: [#] ip -4 rule delete table main suppress_prefixlength 0
Mai 12 23:04:00 Raspap wg-quick[1734]: [#] ip link delete dev wg0
Mai 12 23:04:00 Raspap wg-quick[1734]: [#] resolvconf -d tun.wg0 -f
Mai 12 23:04:00 Raspap wg-quick[1734]: [#] nft -f /dev/fd/63
Mai 12 23:04:00 Raspap wg-quick[1734]: [#] iptables -D FORWARD -i wlan0 -o wg0 -j ACCEPT; iptables -D FORWARD -i wg0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT; iptables -t nat -D  POSTROUTING -o wg0 -j MASQUERADE
-- Boot 511645708481469ba4c151de17190048 --
Mai 12 23:04:14 Raspap wg-quick[885]: [#] ip link add wg0 type wireguard
Mai 12 23:04:14 Raspap wg-quick[885]: [#] wg setconf wg0 /dev/fd/63
Mai 12 23:04:14 Raspap wg-quick[885]: [#] ip -4 address add 192.168.178.203/24 dev wg0
Mai 12 23:04:14 Raspap wg-quick[885]: [#] ip link set mtu 1420 up dev wg0
Mai 12 23:04:14 Raspap wg-quick[936]: [#] resolvconf -a tun.wg0 -m 0 -x
Mai 12 23:04:14 Raspap wg-quick[885]: [#] wg set wg0 fwmark 51820
Mai 12 23:04:14 Raspap wg-quick[885]: [#] ip -4 route add 0.0.0.0/0 dev wg0 table 51820
Mai 12 23:04:14 Raspap wg-quick[885]: [#] ip -4 rule add not fwmark 51820 table 51820
Mai 12 23:04:14 Raspap wg-quick[885]: [#] ip -4 rule add table main suppress_prefixlength 0
Mai 12 23:04:14 Raspap wg-quick[885]: [#] sysctl -q net.ipv4.conf.all.src_valid_mark=1
Mai 12 23:04:14 Raspap wg-quick[885]: [#] nft -f /dev/fd/63
Mai 12 23:04:14 Raspap wg-quick[885]: [#] iptables -A FORWARD -i wlan0 -o wg0 -j ACCEPT; iptables -A FORWARD -i wg0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT; iptables -t nat -A  POSTROUTING -o wg0 -j MASQUERADE
```

billz, 2024-05-14:
@er34zer this will be available in the next release. thanks for reporting!

PR #1586 (2024-05-14): Fix: correctly parse multiple wg DNS entries
Resolves #1585
commit 4175303c (2024-05-14): Fix: correctly parse multiple wg DNS entries, resolves #1585
closed 2024-05-14 by billz