RaspAP/raspap-webgui #1588: [Bug]: Attempting to connect to unsecured network results in failed connection.

kalel681, 2024-05-22:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

External adapter

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

No other software

### Bug description

When attempting to connect to a non-password protected network that doesn't have a captive portal, RaspAP will attempt to connect but refresh the page without connecting. Has been tested on 2.4g, 5g, and wifi 6 networks. 

### Steps to reproduce

Fresh Install
Go To "Wifi Client" in GUI.
"Rescan" to find new network.
"Add" unsecured network (password blank). 
"Connect" to network. 
Page refreshes, network stays in known networks but will not connect. 

### Screenshots

[image]
[image]
[image]
[image]

### Additional context

Debug Log: https://pastebin.com/pycid6PZ

billz, 2024-08-02:
Fixed by #1627

closed 2024-08-02 by billz

kalel681, 2024-08-02:
Just tested and works great. Thanks!