RaspAP/raspap-webgui #1589: [Bug]: DHCP Server Static Lease show nothing

cdr60, 2024-05-26:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 5

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

Yes (specify below)

### Contact details (optional)

ch.derenne@gmail.com

### Bug description

My DHCP server is on Wlan0 interface.
When selecting Wlan0 : I see the dhcp serveurs options properly , in static lease tab, I can add one. It works
Buen when I return to this page tab to add another one, It does not show my previons statis lease. But it is still in dsnmasq.d/090_wlan0.conf file.
The only way to add another static lease is to do it manully by editing 090_wlan0.conf file.

### Steps to reproduce

Default automatic installation (except : no wiregard)
Addblock is disable
Openvpn is on, using a Nordvpn acocunt.
Eth0 is the internet interface (connecting thru openvpn)
Wlan0 is a hotspot and dhcp server
Try to store more than one dhcp static lease for some specifics devices                                                                  (like printer , phone or another PC)

### Screenshots

[image]
[image]
[image]
[image]

billz, 2024-05-29:
Unable to reproduce this with the information provided.

[image]

/etc/dnsmasq.d/090_wlan0.conf 

```
# RaspAP wlan0 configuration
interface=wlan0
dhcp-range=10.3.141.50,10.3.141.254,255.255.255.0,12h
dhcp-host=02:9d:ae:cb:29:6f,10.3.141.248 # iPhone
dhcp-host=dc:a6:32:3d:ff:9b,10.3.141.242 # Galaxy
```

billz, 2024-05-31:
@cdr60 if you've modified the dnsmasq config manually, it's possible that RaspAP is no longer able to parse it. In this case, you can either 1) edit your configuration so it appears like mine, above, or 2) perform a system reset.

closed 2024-07-16 by github-actions[bot]