RaspAP/raspap-webgui #1593: [Bug]: 10x performance drop in network using raspap

frankozland, 2024-06-10:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

frankoz95967943@gmail.com

### Bug description

Topology - (fibre) -> vn router ->(ethernet) -> asus router configured as an access point

        

Scenario 1 - linux manjaro desktop connected directly to asus router via ethernet cable
                     (fibre) -> vn router ->(ethernet) -> asus router configured as an access point ->(ethernet)->linux manjaro desktop
Scenario 2 - linux manjaro desktop connected via wifi to RaspAP
                     (fibre) -> vn router ->(ethernet) -> asus router configured as an access point ->(ethernet)->raspberry pi running raspap -> wifi (5g) -> linux manjaro desktop
Scenario 3 - linux manjaro desktop connected via wifi to to asus router configured as an access point (wifi 5ghz)

RaspAP config - ad blocking is off, its set up with a fresh install, updated os. No RaspAP api, no vpns except openvpn which is disabled.

I've set the country code. there are no dongles or any other hardware installed.

When i run scenario 1 - performance is 285mbps down, 282mbps up
When i run scenario 2 - performance is 25.61down, 21.22 up - and its connected 5ghz
When i run scenario 3 - performance is 282mbps down, 66.4mbps up

So the question i have - what is causing the 10x slowdown on the raspberry?
Would a second eth device - instead of wifi - help?

The device is a raspberry pi 4b  4gb.
[image]

here is the ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.2  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 xxxxx  prefixlen 64  scopeid 0x0<global>
        inet6 xxxxx  prefixlen 64  scopeid 0x20<link>
        ether xxxxx  txqueuelen 1000  (Ethernet)
        RX packets 452785  bytes 479511354 (457.2 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 432398  bytes 324271941 (309.2 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 1148  bytes 106828 (104.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1148  bytes 106828 (104.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet xxxxx  netmask 255.255.255.0  broadcast 10.3.141.255
        inet6 fxxxxxx  prefixlen 64  scopeid 0x20<link>
        ether xxxxx  txqueuelen 1000  (Ethernet)
        RX packets 237201  bytes 151939142 (144.9 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 277203  bytes 299600264 (285.7 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

[image]
[image]
[image]

### Steps to reproduce

install fresh, follow raspap quick install instructions

### Relevant log output

```shell
hostap log

wlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Configuration file: /etc/hostapd/hostapd.conf
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Using interface wlan0 with hwaddr xxxxx and ssid "raspi-webgui"
wlan0: interface state COUNTRY_UPDATE->ENABLED
wlan0: AP-ENABLED 
wlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Configuration file: /etc/hostapd/hostapd.conf
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Using interface wlan0 with hwaddr dc:a6:32:5e:6a:05 and ssid "raspi-webgui"
wlan0: interface state COUNTRY_UPDATE->ENABLED
wlan0: AP-ENABLED 
wlan0: STA xxxxx IEEE 802.11: associated
wlan0: AP-STA-CONNECTED xxxxx
wlan0: STA xxxxx RADIUS: starting accounting session 03BA94B7357885A8
wlan0: STA xxxxx WPA: pairwise key handshake completed (RSN)
wlan0: STA xxxxx IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED xxxxx
wlan0: STA xxxxx IEEE 802.11: disassociated
```

frankozland, 2024-06-10:
one more thing - on the raspberry pi itself - if i run speedtest-cli  - performance is excpetionally good and matches the hi speed performance of scenario 1 and scenario 3

speedtest-cli 
Retrieving speedtest.net configuration...
Testing from VNPT (xxx.xxx.xxx.xxx)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by MobiFone (Ho Chi Minh) [9.17 km]: 5.251 ms
Testing download speed................................................................................
**Download: 285.58 Mbit/s**
Testing upload speed......................................................................................................
**Upload: 269.14 Mbit/s**
dave@raspberrypirouter:~ $

billz, 2024-06-10:
https://docs.raspap.com/faq/#433ac

billz, 2024-06-10:
This is a limitation of the hardware, not of RaspAP itself.

closed 2024-06-10 by billz