RaspAP/raspap-webgui #1595: [Bug]: Cannot select eth1 interface  for eth1 to eth0 bridge

frankozland, 2024-06-11:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

frankoz95967943@gmail.com

### Bug description

I think there is an issue with the OS network stack and thats the purpose of the tests.
Totally ok - just want to get to bottom of it.
Its not just raspap - its also expressvpn - without raspap even installed - performance is abysmal.
Raspap has a similar issue - and because you are using cli so heavy/linux directly wrapped with php - your sw is perfect for isolating the layer thats having issues.

Sorry to be pest on all this - i really only trying to help the community.

On raspap - i only have raspap installed with the openvpn option installed but openvpn is disabled.
no ad block, no other cli or raspap api installed. expressvpn was removed and purged.

I have eth0 and eth1 and wlan0 (built in)

I confirmed both eth0 and eth1 works on raspberry.

The topology now is:
(fiber) -> router -> asus router acting as Access point -> (ethernet) -> eth0 on raspberry pi running raspap -> eth1 dongle -> (ethernet) -> linux desktop running manjaro linux

on the config for hotspot there is an option to select eth0, eth1, lo, wlan0. I shut down hotspot and selected eth1.
I attempted to "save settings"
This option is disabled.

So then i tried to start the hotspot again. It defaulted to wlan0.
so then i did a soft kill in linux to disable the wlan0
sudo rfkill block wifi
and went back to raspap - and tried to set to eth1.
it didnt work - hotspot would not start because it the wlan0 was disabled.

rfkill: WLAN soft blocked
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Using interface wlan0 with hwaddr dc:a6:32:5e:6a:05 and ssid "raspi-webgui"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state COUNTRY_UPDATE->DISABLED
wlan0: AP-DISABLED
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING

So digging in a little - if user doesnt select a wifi device, the save button is disabled.
I think its in the .js code thats doin this - im not sure if thats intentional

its in custom.js starting on line 533 - eth1 isnt a wifi device - so no channel is selected and then the save settings button is disabled.

if (hw_mode === 'a') {
selectableChannels = data.filter(item => item.MHz.toString().startsWith('5'));
} else if (hw_mode !== 'ac') {
selectableChannels = data.filter(item => item.MHz.toString().startsWith('24'));
} else if (hw_mode === 'b') {
selectableChannels = data.filter(item => item.MHz.toString().startsWith('24'));
} else if (hw_mode === 'ac') {
selectableChannels = data.filter(item => item.MHz.toString().startsWith('5'));
}

    // If selected channel doeesn't exist in allowed channels, set default or null (unsupported)
    if (!selectableChannels.find(item => item.Channel === selected)) {
        if (selectableChannels.length === 0) {
            selectableChannels[0] = { Channel: null };
        } else {
            defaultChannel = selectableChannels[0].Channel;
            selected = defaultChannel
        }
    }

    // Set channel select with available values
    channel_select.empty();
    if (selectableChannels[0].Channel === null) {
        channel_select.append($("<option></option>").attr("value", "").text("---"));
        channel_select.prop("disabled", true);
        **btn_save.prop("disabled", true);**    // <---------------save button disabled here because selected device isnt wifi
    } else {
        channel_select.prop("disabled", false);
        btn_save.prop("disabled", false);
        $.each(selectableChannels, function(key,value) {
            channel_select.append($("<option></option>").attr("value", value.Channel).text(value.Channel));
        });
        channel_select.val(selected);
    }

### Steps to reproduce

[image]

billz, 2024-06-11:
Duplicate of https://github.com/RaspAP/raspap-webgui/discussions/1591. Think this is better addressed in your original thread, so closing this (which is less a bug report and more a feature request). Please be patient, I'll get to it. Thanks

billz, 2024-06-12:
@frankozland I've tagged this as a feature request with an explanation in your discussion thread. thanks!

closed 2024-06-12 by billz