RaspAP/raspap-webgui #1597: Kill switch

frankozland, 2024-06-16:
## Is your feature request related to a problem? 
If the hosted vpn goes down i really want all traffic blocked and a status on RaspAP dashboard to reflect the broken pipe.
I think an option to the user to kill traffic (or a switch in the interface) to enable action on openvpn or client vpn pages.
It'd be nice if a message was logged somewhere that the network was killed with a date time so i can work with vpn vendor about stability/network channel.

## Describe the solution you'd like
Stop network traffic if vpn down

## Describe alternatives you've considered
openvpn up/down scripts

## Additional context

Would this do it? Seems trivial to add to install script?

The following "kill switch" will prevent all clients on the private network from accessing the WAN should the VPN drop.

Code:
WAN_IF="$(ip route | awk '/^default/{print $NF}')"
iptables -I FORWARD -i br0 -o $WAN_IF -m state --state NEW -j REJECT --reject-with icmp-host-prohibited
iptables -I FORWARD -i br0 -p tcp -o $WAN_IF -m state --state NEW -j REJECT --reject-with tcp-reset
</quote>

billz, 2024-06-16:
Something like this? 😉 https://docs.raspap.com/wireguard/#kill-switch

frankozland, 2024-06-16:
yes sir - but for openvpn - rn if openvpn dies? raspap still allows traffic

i found this - i think its close....
https://github.com/renapoliveira/killswitch-for-openvpn/blob/master/killswitch.sh

The only problem is expressvpn doesnt advertise the protocol in the conf file

All traffic must stop on openvpn disconnect/pause/timeout/etc

frankozland, 2024-06-16:
in this script to get config - all of this works except getting the protocol - expressvpn uses UDP for openvpn but its listed nowhere in the conf.

function config() {	
	#Get the default network interface
	echo "Detecting your default network interface..."
	INTERFACE=`ip addr | grep "state UP" | cut -d ":" -f 2 | head -n 1`
	echo "Using "$INTERFACE

	TUNNEL=tun0
	echo "Using interface "$TUNNEL " for VPN, change the script if you need another one."	

	#Get the VPN IP, PORT and PROTOCOL from the VPN file
	echo "Detecting your VPN server address..."
	IP=`cat "$VPN_FILE" | grep "remote " | awk '{print $2}'`
	echo "Using IP "$IP

	echo "Detecting your VPN port..."
	PORT=`cat "$VPN_FILE" | grep "remote " | awk '{print $3}'`
	echo "Using port "$PORT

	echo "Detecting your VPN protocol..."
	PROTOCOL=`cat "$VPN_FILE" | grep "proto " | awk '{print $2}'`
	echo "Using protocol "$PROTOCOL
}

frankozland, 2024-06-16:
Digging a little more - theres quite a bit that can be done outside of raspap which might solve my use case.
I can detect with a bash script if openvpn process is or is not there. If its not there, i shut down hotspot.
I can also monitor journalctl - openvpn handles signals "usr2" which will log status if its operational - and i can parse them out of journal.

And openvpn will log issues which can be trapped and handled with a bash script.

journalctl -f _COMM=openvpn -n0 
Strings that can be trapped:
"TUN/TAP device tun0 opened" --tunnel opened
"Initialization Sequence Completed" -- openvpn is up
"Closing TUN/TAP interface" -- need to hard stop raspap
"FRAG TTL","TLS: soft reset","Restart" -- warn user connection is shaky; count the strings and display
"SIGTERM" -- hard stop on raspap

--get openvpn process
openvpn_ps=$(ps -f -U root | grep openvpn | awk {'print $2'})
if not there, make sure raspap hotspot is down

--periodically get status; sending USR2 signal makes openvpn dump its status to the journal
sudo kill -s USR2 $openvpn_ps
--this command will dump this into the log which can be trapped
Jun 16 15:15:02 raspberryrouter openvpn[20351]: OpenVPN STATISTICS
Jun 16 15:15:02 raspberryrouter openvpn[20351]: Updated,2024-06-16 15:15:02
Jun 16 15:15:02 raspberryrouter openvpn[20351]: TUN/TAP read bytes,5991
Jun 16 15:15:02 raspberryrouter openvpn[20351]: TUN/TAP write bytes,1601
Jun 16 15:15:02 raspberryrouter openvpn[20351]: TCP/UDP read bytes,5620
Jun 16 15:15:02 raspberryrouter openvpn[20351]: TCP/UDP write bytes,11012
Jun 16 15:15:02 raspberryrouter openvpn[20351]: Auth read bytes,1601
Jun 16 15:15:02 raspberryrouter openvpn[20351]: pre-compress bytes,0
Jun 16 15:15:02 raspberryrouter openvpn[20351]: post-compress bytes,0
Jun 16 15:15:02 raspberryrouter openvpn[20351]: pre-decompress bytes,0
Jun 16 15:15:02 raspberryrouter openvpn[20351]: post-decompress bytes,0
Jun 16 15:15:02 raspberryrouter openvpn[20351]: END

logic - monitor openvpn
if process is not there, shut down hostap
if detect "Closing TUN/TAP interface" kill raspap process immediately
if detect "SIGTERM" on openvpn kill raspap process immediately

I might be able to solve this outside of raspap for now.

In researching this is an ongoing issue with openvpn and few have a decent solution (besides doing a iptables solution) - there might be a 2 part process to this - to monitor for activity, if up - add iptables.  if going down - kill hotspot, then remove ip tables.  

Run the whole thing in a background process.

I think that might be the tightest solution.

closed 2024-08-01 by github-actions[bot]

wolfiex, 2025-01-19:
> Something like this? 😉 https://docs.raspap.com/wireguard/#kill-switch

+1 for something like this for OpenVPN - have there been any updates with this?

billz, 2025-01-19:
@wolfiex user contributed iptables rules are [available here](https://github.com/RaspAP/raspap-webgui/issues/1413#issuecomment-2586505755). these haven't been integrated yet as it requires a substantial testing effort, as well as integration with RaspAP's firewall. if you can contribute to either of these efforts, this will speed things along.