RaspAP/raspap-webgui #1598: [Bug]: RestAPI Fail to Start

gwashark, 2024-06-20:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bullseye

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

External adapter

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

No other software

### Bug description

I installed RaspAp using the quick setup and I said yes to the `Rest API`. But whenever I boot or try and initiate it myself, it always returns `code=exited, status=217/USER`.

### Steps to reproduce

1. Start up the Raspberry Pi.
2. Login to the dashboard.
3. Go to `Rest API` settings.
4. Click `Start RestAPI Service` button.

### Screenshots

[image]

### Additional context

I tried setting the api key and attempted to start it multiple times but it kept saying the same error. 

### Relevant log output

```shell
Loaded: loaded (/lib/systemd/system/restapi.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2024-06-19 23:19:26 EDT; 93ms ago
    Process: 3016 ExecStart=/usr/bin/python3 -m uvicorn main:app --host 0.0.0.0 --port 8081 (code=exited, status=217/USER)
   Main PID: 3016 (code=exited, status=217/USER)
        CPU: 0
```

gwashark, 2024-06-20:
I’ve made a mistake. I am running, the `Bookworm` version.

Here is what I’m working with.
```
PI REVISION: Raspberry Pi 4 Model B Rev 1.5
OS: Debian GNU/Linux 12 (bookworm)
KERNEL: 6.6.31+rpt-rpi-v8
```

poldy42, 2024-06-20:
Good morning,
are you working with a user other than the default "pi" user ? The systemd error message almost sounds like it. In this case, you would have to adjust the "User" part in the /lib/systemd/system/restapi.service file! Otherwise the system will not find the files installed by pip....
Cheers - Michael

gwashark, 2024-06-20:
Whoops! Looks like I didn’t get into the docs more.😅

That is the solution to my problem.

closed 2024-06-20 by gwashark

Bluemethyst, 2025-01-29:
Im getting this same error when running on Docker on my Pi 5