RaspAP/raspap-webgui #1600: [Bug]: hostapd erros in journal

frankozland, 2024-06-22:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

frankoz95967943@gmail.com

### Bug description

Repeatable problem.
Out of the box quick install - set wlan country - no other config changes.

In raspap - changed to 5ghz

As soon as i did that the raspberry network dropped - even ssh wouldnt connect.

Log file began to fill with 
Jun 22 03:57:06 raspberrypirouter kernel: ieee80211 phy0: brcmf_run_escan: error (-52)
Jun 22 03:57:06 raspberrypirouter kernel: ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
Jun 22 03:57:08 raspberrypirouter kernel: ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=53282, -52

until i killed hostapd service.

going back to 2.4ghz channel 1 everything returned to normal, no errors in log.

[image]

[image]

### Steps to reproduce

see above

billz, 2024-06-22:
This is a very common error thrown by hostapd. Suggest you google first. Top results: https://forums.raspberrypi.com/viewtopic.php?t=249750
https://github.com/raspberrypi/linux/issues/3849

If not done so already, disabling power management may help. RaspAP is a magnet for reports like this, when often the issue exists at the level of the kernel or firmware. Please bear in mind that this project is, fundamentally, a configuration manager. It isn't magic and cannot work miracles.

billz, 2024-06-22:
I carved out some time and upgraded to the latest kernel. I'm able to reproduce this.
```
$ uname -a
Linux rpi5 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
```

```
$ sudo systemctl restart hostapd.service; journalctl -f
Jun 22 11:42:16 rpi5 kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
Jun 22 11:42:16 rpi5 kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
Jun 22 11:42:16 rpi5 kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
Jun 22 11:42:16 rpi5 kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
Jun 22 11:42:19 rpi5 kernel: ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=53282, -52
```

Appears it has already been reported https://github.com/raspberrypi/linux/issues/6049

frankozland, 2024-06-22:
Incredible support and response.

Thank you mr. billz - im ok with downgraded version - i'll monitor if they fix the OS.

Im guessin anyone who follows instructions to install raspap w/ apt full-upgrade will run into the exact same scenario.
I dont know when the network stack was broken or max OS version its not.

frankozland, 2024-06-22:
https://wireless.wiki.kernel.org/en/users/documentation/acs

hostap has the option to not specify a channel and it will automatically select channel for you - i dont know if that will fix it, but there seems to be work here on not specifying any channel when setting up hotspot

billz, 2024-06-22:
> Im guessin anyone who follows instructions to install raspap w/ apt full-upgrade will run into the exact same scenario.
I dont know when the network stack was broken or max OS version its not.

News to me also. As evidenced by the forum posts and previous issues, this isn't the first time a kernel update hosed brcmfmac, nor will it be the last.

feh123, 2024-07-11:
> Incredible support and response.
>
> Thank you mr. billz - im ok with downgraded version - i'll monitor if they fix the OS.
>
> Im guessin anyone who follows instructions to install raspap w/ apt [...]

Hi I have a RPI 4 running with a second wifi adapter and I get a very similar error - brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52 when I install bookworm. I did try using sudo apt-get upgrade rather than full-upgrade but I still saw the same error and no AP. My RaspAP works fine with Bullseye. I am well out of my depth with this so not sure exactly what is happening! My situation may not be relevant to your problem. I will be interested to see if a new OS is issued with a fix for you.

nima-1102, 2024-08-11:
+1 Latest firmware, all updates installed, Raspberry Pi Os Lite

billz, 2024-08-11:
@nima-1102 this is not a RaspAP bug. Refer to the pinned discussion.

closed 2024-08-11 by billz