RaspAP/raspap-webgui #1601: [Bug]: wg status shows down in raspap but its running and connected on os

frankozland, 2024-06-22:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 5

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

Yes (specify below)

### Contact details (optional)

frankoz95967943@gmail.com

### Bug description

configured raspap wg using a config file

upload and start wg thru raspap.

raspap shows "wg down"

os shows wg is running
[image]

[image]

### Steps to reproduce

[raspap_debug(1).log](https://github.com/user-attachments/files/15935826/raspap_debug.1.log)

billz, 2024-06-22:
Execute `pidof wg-crypt-wg0 | wc -l` in the shell.

lyra-gh, 2024-07-07:
Same problem here:
```
$ sudo wg show
interface: wg0
  public key: LbR9UOkG0Pw6/3yFqCUBpBO5CkDgEwENAN5vxuQjLEg=
  private key: (hidden)
  listening port: 37845
  fwmark: 0xca6c

peer: vPQ8lIBL9nkClRjzlQisz7XZV16ujf1MvpQX6uzbPng=
  preshared key: (hidden)
  endpoint: 91.193.5.50:251
  allowed ips: 0.0.0.0/0
  latest handshake: 27 seconds ago
  transfer: 23.02 MiB received, 68.27 MiB sent
  persistent keepalive: every 25 seconds
$ pidof wg-crypt-wg0 | wc -l
0
```

PR #1612 (2024-07-07): Update method to determine wg-quick state
Fixes #1601
commit 828a4638 (2024-07-07): Update method to determine wg-quick state, resolves #1601
closed 2024-07-07 by billz