RaspAP/raspap-webgui #1602: [Bug]: raspap doesnt advertise wifi ssid

frankozland, 2024-06-22:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

Yes (specify below)

### Contact details (optional)

frankoz95967943@gmail.com

### Bug description

fresh install, apt update, apt full_upgrade
country code set
wg and openvpn installed, both inactive

fresh reboot
raspap does not advertise ssid
says hostap is up

Jun 22 04:53:32 raspberrypirouter systemd[1]: Started hostapd.service - Access point and authentication server for Wi-Fi and Ethernet.
Jun 22 04:53:39 raspberrypirouter sudo[2071]: www-data : PWD=/var/www/html ; USER=root ; COMMAND=/bin/chmod o+r /tmp/hostapd.log
Jun 22 04:53:48 raspberrypirouter sudo[2203]: www-data : PWD=/var/www/html ; USER=root ; COMMAND=/bin/chmod o+r /tmp/hostapd.log
dave@raspberrypirouter:~ $ cat /tmp/hostapd.log
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: interface state COUNTRY_UPDATE->ENABLED
wlan0: AP-ENABLED 
wlan0: INTERFACE-DISABLED 
wlan0: INTERFACE-ENABLED 
nl80211: kernel reports: key not allowed
Failed to set beacon parameters
wlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: interface state COUNTRY_UPDATE->ENABLED
wlan0: AP-ENABLED 
[... 347 lines cut ...]
             ├─1256 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
             ├─1298 /usr/bin/php-cgi
             ├─1299 /usr/bin/php-cgi
             ├─1300 /usr/bin/php-cgi
             ├─1301 /usr/bin/php-cgi
             ├─1302 /usr/bin/php-cgi
             ├─2555 sh -c "sudo /etc/raspap/system/debuglog.sh -i /var/www/html"
             ├─2556 sudo /etc/raspap/system/debuglog.sh -i /var/www/html
             ├─2557 /bin/bash /etc/raspap/system/debuglog.sh -i /var/www/html
             ├─2681 /bin/bash /etc/raspap/system/debuglog.sh -i /var/www/html
             └─2682 systemctl status lighttpd

Jun 22 04:54:10 raspberrypirouter sudo[2345]: pam_unix(sudo:session): session closed for user root
Jun 22 04:54:10 raspberrypirouter sudo[2348]: www-data : PWD=/var/www/html ; USER=root ; COMMAND=/bin/cat /etc/wireguard/client.conf
Jun 22 04:54:10 raspberrypirouter sudo[2348]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Jun 22 04:54:10 raspberrypirouter lighttpd[2349]: cat: /etc/wireguard/client.conf: No such file or directory
Jun 22 04:54:10 raspberrypirouter sudo[2348]: pam_unix(sudo:session): session closed for user root
Jun 22 04:54:12 raspberrypirouter sudo[2356]: www-data : PWD=/var/www/html ; USER=root ; COMMAND=/bin/chmod o+r /tmp/wireguard.log
Jun 22 04:54:12 raspberrypirouter sudo[2356]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Jun 22 04:54:12 raspberrypirouter sudo[2356]: pam_unix(sudo:session): session closed for user root
Jun 22 04:57:47 raspberrypirouter sudo[2556]: www-data : PWD=/var/www/html/ajax/system ; USER=root ; COMMAND=/etc/raspap/system/debuglog.sh -i /var/www/html
Jun 22 04:57:47 raspberrypirouter sudo[2556]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)

RaspAP debug log generation complete.
```

billz, 2024-06-22:
@frankozland if hostapd is throwing errors like you reported in #1600, the service may be active but has entered a failed state. you'll need to get to the root of the other issue first.

closed 2024-06-22 by billz

frankozland, 2024-06-22:
its a fresh install - same issue on 2 different identical raspberries.
im going to downgrade os and raspap to prior version

frankozland, 2024-06-22:
reboots dont fix it, moving to duplicate raspberry pi doesnt fix it. - its virgin install. there is something going on w/ os is my guess- there is a bug in this release at network stack.

frankozland, 2024-06-22:
same hardware, downgraded OS and raspap. 
now everything works.

the OS has a bug.

billz, 2024-06-22:
It is indeed. See https://github.com/RaspAP/raspap-webgui/issues/1600#issuecomment-2183963931

spencer1573, 2024-07-29:
i used the latest raspberry pi os with full-upgrade and couldnt get it working. so it looks like this is still an issue. i thought i followed the instructions correctly. i'm using a 
pi 5. 
raspberry pi os lite 64 bit
did the `full-upgrade` command

i dont know if i missed something. but its not working how i think it should.