RaspAP/raspap-webgui #1603: Allow for group id specified in openvpn config/enable kill switch for openvpn

frankozland, 2024-06-22:
## Is your feature request related to a problem? 

Very few other solutions exist that have a preconfigured easy to use way to have a reliable killswitch for openvpn.
I get that wireguard has a kill switch, as do some of the other vendors - but not everyone can use them for whatever reason.

The goal is to enable a kill switch for openvpn.

I think this will work:

raspap uses services file to launch openvpn (in openvpn includes):

elseif (isset($_POST['StartOpenVPN'])) {
            $status->addMessage('Attempting to start OpenVPN', 'info');
            exec('sudo /bin/systemctl start openvpn-client@client', $return);

But there is no group specified (groups make firewall config a LOT simpler for openvpn)

## Describe the solution you'd like

step 1 - add new group on OS
sudo groupadd raspap_openvpn

step 2 - enable ip forwarding
sudo  sysctl -w net.ipv4.ip_forward=1
sudo sysctl -p

step 3 - alter service to use this group
add group to in services file on openvpn launch
ie 
ExecStart=/usr/sbin/openvpn --suppress-timestamps --nobind **--group raspap_openvpn** --config %i.conf

this will automatically set the group at launch of openvpn.

[Unit]
Description=OpenVPN tunnel for %I
After=network-online.target
Wants=network-online.target
Documentation=man:openvpn(8)
Documentation=https://community.openvpn.net/openvpn/wiki/Openvpn24ManPage
Documentation=https://community.openvpn.net/openvpn/wiki/HOWTO

[Service]
Type=notify
PrivateTmp=true
WorkingDirectory=/etc/openvpn/client
ExecStart=/usr/sbin/openvpn --suppress-timestamps --nobind **--group raspap_openvpn** --config %i.conf

or 

rewrite config.ovpn to add a new line at top of config when moving it into place
**group raspap_openvpn**
at the top of the file

## Describe alternatives you've considered
routing tables and kludgy solutions for a kill switch

## Additional context
Nice to have - once group is created for kill switch - offer kill switch on/off
and add these rules to firewall

# Local traffic is allowed
iptables -A INPUT  -i lo -j ACCEPT
iptables -A OUTPUT -o lo -j ACCEPT
# Real interfaces are allowed only for tunnel group
iptables -I OUTPUT -o wlan+ -m owner \! --gid-owner raspap_openvpn -j REJECT
iptables -I OUTPUT -o eth+  -m owner \! --gid-owner raspap_openvpn -j REJECT
iptables -t nat -A POSTROUTING -o tun0 -j MASQUERADE

billz, 2024-06-22:
@frankozland is this not identical to #1597 ?

frankozland, 2024-06-22:
related - specific to expressvpn (i could only use their openvpn reliably)
they now have a saturation problem i think.

moving over to protonvpn - very very fast, i'd like to continue to use openvpn - the target state is to drop everything if openvpn goes down.

I did see insiders for wireguard kill switch - i dont know if that would work in my case; openvpn is giving me very good performance and its been reliable for years.

i gave up on iptables and moved to ufw.

sudo apt-get install ufw

then the rules:
#allow local network access to raspberry
ufw allow to 192.168.1.0/24
ufw allow out to 192.168.1.0/24
ufw allow from 192.168.1.0/24
#allow ssh
ufw allow 22/tcp
ufw allow out 22/tcp
#allow dns
ufw allow out 53
ufw allow 53
ufw allow 5901:5910/tcp
#allow tunnel
ufw allow out on tun0
ufw allow in on tun0
#allow lo
ufw allow in on lo
ufw allow out on lo
#this is the static port configured below for openvpn
ufw allow out 1198/udp
ufw allow 1198/udp
#allow access to port 80 to connect to raspap even if tunnel is down
ufw allow from 192.168.1.0/24 to any port 80
#then turn on all the rules
ufw enable

and it required a change to openvpn client demon (else need to rewrite every opvn conf file):

/lib/systemd/system/openvpn-client@.service openvpn-client@.service
and change this:
ExecStart=/usr/sbin/openvpn --suppress-timestamps --nobind   --config %i.conf
to this:
ExecStart=/usr/sbin/openvpn --suppress-timestamps **--local 0.0.0.0 --lport 1198** --config %i.conf

With this change, and ufw - openvpn now has a killswitch ... i think....

by changing the option on the command line, I no longer need to worry about changing every opvn conf i download.
im just specifying the local ip and the local port to be consistent - this will only work if you only have 1 openvpn client running on the device.

with these rules - when openvpn goes down now, i still have access to raspap thru the firewall and ssh server. All local nearby lan can connect to the raspberry.  All other traffic is blocked - i think (im still testing).

I dont know if there'd be interest in this for user base - i know from this seat there is - anyone using openvpn is at risk of leak if openvpn goes down.  icmp is still allowed and im guessin if someone wanted to discover the channel with openvpn down theres a trigger to start sniffin around.

These firewall rules might work for other CLI's - expressvpn,proton, etc - they all construct a tunnel, but im not sure about which port they are communicating on or if there is a way to interrogate that with their cli (or if its worth it)

This is what i used as a guide:
https://gist.github.com/Necklaces/18b68e80bf929ef99312b2d90d0cded2

frankozland, 2024-06-22:
one other caviet to this approach - the downloaded ovpn for openvpn seem to include in the client.conf file a "nobind" parameter - which conflicts with the commandline "bind" parameter. so after downloading the file has to be modified and that line commented out.

closed 2024-08-06 by github-actions[bot]