RaspAP/raspap-webgui #1611: [Bug]: Unable to Obtain IP in AP Mode

velison1-mbed, 2024-07-05:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(32-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi Zero W

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

No other software

### Bug description

Dear RaspAP team,

I am using Raspberry Pi Zero W integrated into a custom PCB with Raspberry Pi OS (released on March 15, 2024) and RaspAP version 3.1.3. After completing the RaspAP setup, I can see the `raspi-webgui` SSID in Access Point mode, but I am unable to obtain an IP address when attempting to connect to it.

Additionally, the device has an Ethernet interface that I use regularly. However, I recently tried connecting via Wi-Fi and noticed that, despite the network details being correctly added to `wpa_supplicant.conf`, the device fails to connect to the specified network. Upon further investigation, I found that this issue might be related to NetworkManager.

In the release notes for the Raspberry Pi OS version from October 10, 2023, it was mentioned that:
* NetworkManager is now used instead of dhcpcd as the networking interface; various changes were made to the networking plugin to support this.

Previously, I had encountered an issue with the absence of the `dhcpcd.service` while using this version, which was resolved as discussed in this [issue]. Currently, the `dhcpcd` service is available, and the `raspapd` service starts without any apparent errors, but I still face the problem of not being able to connect to the `raspi-webgui` network.

I have tried both the quick installation and manual installation methods, but the issue persists.

Do you plan to integrate RaspAP with NetworkManager in the future? What do you recommend for using RaspAP with the latest versions of Raspberry Pi OS? Should we continue using older versions of the OS for now to avoid this issue?

Best regards,

Veli Can

### Steps to reproduce

Here are the steps I followed for a clean installation, as per the RaspAP documentation:

sudo apt-get update
sudo apt-get full-upgrade
sudo reboot

curl -sL https://install.raspap.com | bash

### Relevant log output

```shell
Jul 05 11:57:59 raspberrypi bash[4433]: Changing RaspAP Daemon --interface to uap0
Jul 05 11:57:59 raspberrypi bash[4433]: Starting network services...
Jul 05 11:58:07 raspberrypi bash[4433]: RaspAP service start DONE
Jul 05 11:58:07 raspberrypi systemd[1]: raspapd.service: Deactivated successfully.
Jul 05 11:58:07 raspberrypi systemd[1]: Finished raspapd.service - RaspAP Service Daemon.
```

billz, 2024-07-05:
You may be running into a known kernel bug. Refer to #1600 and [this discussion](https://github.com/RaspAP/raspap-webgui/discussions/1606#discussioncomment-9874248). Impossible to tell however as there is no dhcp log associated with your issue. For the latter, see [this FAQ](https://docs.raspap.com/faq/#noip). Please spend some time in the docs and follow [these guidelines](https://docs.raspap.com/issues/#guidelines) before creating new issues. Thanks

velison1-mbed, 2024-07-05:
These are logs from journalctl

Jul 05 17:00:28 raspberrypi hostapd[986]: wlan0: STA 26:05:e9:98:ec:6d IEEE 802.11: associated
Jul 05 17:00:28 raspberrypi hostapd[986]: wlan0: STA 26:05:e9:98:ec:6d RADIUS: starting accounting session 7D53F70E07BC1E80
Jul 05 17:00:28 raspberrypi hostapd[986]: wlan0: STA 26:05:e9:98:ec:6d WPA: pairwise key handshake completed (RSN)
Jul 05 17:00:47 raspberrypi hostapd[986]: wlan0: STA 26:05:e9:98:ec:6d IEEE 802.11: disassociated
Jul 05 17:00:47 raspberrypi hostapd[986]: wlan0: STA 26:05:e9:98:ec:6d IEEE 802.11: disassociated
Jul 05 17:01:08 raspberrypi dhcpcd[1442]: ps_sendcmdmsg: Connection refused
Jul 05 17:01:08 raspberrypi dhcpcd[1442]: ps_inet_recvra: Connection refused
Jul 05 17:01:21 raspberrypi systemd[1]: dhcpcd.service: State 'stop-sigterm' timed out. Killing.
Jul 05 17:01:21 raspberrypi systemd[1]: dhcpcd.service: Killing process 1442 (dhcpcd) with signal SIGKILL.
Jul 05 17:01:21 raspberrypi systemd[1]: dhcpcd.service: Killing process 1443 (dhcpcd) with signal SIGKILL.
Jul 05 17:01:21 raspberrypi systemd[1]: dhcpcd.service: Failed with result 'signal'.
Jul 05 17:01:21 raspberrypi systemd[1]: dhcpcd.service: Scheduled restart job, restart counter is at 28.
Jul 05 17:01:21 raspberrypi systemd[1]: Stopped dhcpcd.service - DHCP Client Daemon.
Jul 05 17:01:21 raspberrypi systemd[1]: Starting dhcpcd.service - DHCP Client Daemon...
Jul 05 17:01:21 raspberrypi dhcpcd[1452]: main: control_open: Connection refused
Jul 05 17:01:21 raspberrypi dhcpcd[1452]: main: control_open: Connection refused
Jul 05 17:01:21 raspberrypi dhcpcd[1452]: dhcpcd-9.4.1 starting
Jul 05 17:01:21 raspberrypi dhcpcd[1455]: dev: loaded udev
Jul 05 17:01:21 raspberrypi dhcpcd[1455]: DUID 00:01:00:01:2e:1a:ae:cb:b8:27:eb:7f:2d:68
Jul 05 17:01:21 raspberrypi systemd[1]: Started dhcpcd.service - DHCP Client Daemon.
Jul 05 17:01:21 raspberrypi dhcpcd[1455]: wlan0: connected to Access Point:
Jul 05 17:01:22 raspberrypi dhcpcd[1455]: eth0: IAID eb:7f:2d:68
Jul 05 17:01:21 raspberrypi audit[1454]: SECCOMP auid=4294967295 uid=106 gid=65534 ses=4294967295 pid=1454 comm="dhcpcd" exe="/usr/sbin/dhcpcd" sig=31 arch=40000028 syscall=384 compat=0 ip=0xb6e6d330 code=0x0
Jul 05 17:01:22 raspberrypi kernel: audit: type=1326 audit(1720188081.999:31): auid=4294967295 uid=106 gid=65534 ses=4294967295 pid=1454 comm="dhcpcd" exe="/usr/sbin/dhcpcd" sig=31 arch=40000028 syscall=384 compat=0 ip=0xb6e6d330 code=0x0
Jul 05 17:01:22 raspberrypi dhcpcd[1455]: ps_ctl_listen: read: Success
Jul 05 17:01:22 raspberrypi dhcpcd[1455]: ps_ctl_recv: read: Success
Jul 05 17:01:22 raspberrypi systemd[1]: dhcpcd.service: Main process exited, code=killed, status=31/SYS

billz, 2024-07-05:
There are any number of possible causes for this. There's nothing I, or anyone else, can do to help you determine this without a) dhcp logs (not journalctl) and b) your dhcpcd.conf. You checked the box indicating that you've generated a debug log and done a self-diagnosis... but did you actually do these things? Re-read the issue policy and decide for yourself if this is actually a bug, a configuration error on your part or something else. This is not a troubleshooting forum.

velison1-mbed, 2024-07-08:
I opened the topic in a way that was not clear in the wrong place. I'm sorry for taking your time. The topic can be closed. I will try to work on it more and write more clearly. Thanks for your help.

Best regards,

Veli Can

billz, 2024-07-08:
No problem. I invite you to post a new discussion where I (or other users of RaspAP) can assist you. This is the preferred forum for these things. Thanks.

velison1-mbed, 2024-07-08:
Thanks a lot.

closed 2024-08-22 by github-actions[bot]