RaspAP/raspap-webgui #1624: Enable BBR by default for speed improvement

frankozland, 2024-07-19:
## Is your feature request related to a problem? 
Default TCP scheduler in Rasperry pi is pfifo_fast. this is great for edge use case of single user process space.
RaspAP is more of a router type use case.

## Describe the solution you'd like
On install - 

sysctl net.core.default_qdisc=fq
sysctl net.ipv4.tcp_congestion_control=bbr
sysctl -p

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
https://www.slideshare.net/slideshow/tcp-and-bbr/89242607#30
[image]

frankozland, 2024-07-19:
let me run a few more tests

billz, 2024-08-03:
Nice idea, I'd like to see the results of your tests. Thanks for the suggestion

billz, 2024-08-18:
@frankozland this option will be available in the next general release. thanks!

closed 2024-08-18 by billz