RaspAP/raspap-webgui #1625: [Bug]:

Drizzt321, 2024-07-20:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(32-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi Zero W

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

drizzt321@gmail.com

### Bug description

I'm trying to install on a RPi Zero W that I've had laying around. I have an 64GB mSD card, did a fresh install of 32-bit Bookworm Lite Raspian. Updated/full-upgrade, and then followed https://docs.raspap.com/ap-sta/#installation instructions for RPi Zero W installation. 

After the install via the Quick Install script, the `uap0` interface was there, but down. When I would try and bring it up via the `Start uap0` button on the Dashboard page. That wouldn't work, in fact got an error in `journalctl` log saying no permissions. Even when I tried `ip link set uap0 up` as root, it kept showing as down. And no WiFi network with the chosen network name appears to clients. 

### Steps to reproduce

Unsure if anything specific, but fresh Raspian Lite 32-bit on RPi Zero W v1.1, then follow install instructions at https://docs.raspap.com/ap-sta/#installation including _not_ rebooting until after I had tried everything on that installation page, even checked 

### Screenshots

My `/etc/raspap/hostapd.ini` is

```
WifiInterface = uap0
LogEnable = 1
WifiAPEnable = 1
BridgedEnable = 0
WifiManaged = wlan0
```

### Relevant log output

```shell
journalctl after hitting Start Hotspot before Reboot:

Jul 19 19:26:28 playapologies sudo[8788]:     root : TTY=pts/1 ; PWD=/root ; USER=root ; COMMAND=/usr/bin/mv /var/www/html/installers/restapi.service /lib/systemd/system/
[... 531 lines cut ...]
             ├─ 8100 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
             ├─ 8723 /usr/bin/php-cgi
             ├─ 8724 /usr/bin/php-cgi
             ├─ 8725 /usr/bin/php-cgi
             ├─ 8726 /usr/bin/php-cgi
             ├─ 8727 /usr/bin/php-cgi
             ├─10176 sh -c "sudo /etc/raspap/system/debuglog.sh -i /var/www/html"
             ├─10177 sudo /etc/raspap/system/debuglog.sh -i /var/www/html
             ├─10178 /bin/bash /etc/raspap/system/debuglog.sh -i /var/www/html
             ├─10301 /bin/bash /etc/raspap/system/debuglog.sh -i /var/www/html
             └─10302 systemctl status lighttpd

Jul 19 19:31:45 playapologies sudo[9926]: pam_unix(sudo:session): session closed for user root
Jul 19 19:32:14 playapologies sudo[10001]: www-data : PWD=/var/www/html ; USER=root ; COMMAND=/bin/chmod o+r /tmp/hostapd.log
Jul 19 19:32:14 playapologies sudo[10001]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Jul 19 19:32:14 playapologies sudo[10001]: pam_unix(sudo:session): session closed for user root
Jul 19 19:32:16 playapologies lighttpd[10081]: Usage: grep [OPTION]... PATTERNS [FILE]...
Jul 19 19:32:16 playapologies lighttpd[10081]: Try 'grep --help' for more information.
Jul 19 19:33:12 playapologies sudo[10177]: www-data : PWD=/var/www/html/ajax/system ; USER=root ; COMMAND=/etc/raspap/system/debuglog.sh -i /var/www/html
Jul 19 19:33:12 playapologies sudo[10177]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Jul 19 19:33:14 playapologies lighttpd[10298]: Unit wg-quick@wg0.service could not be found.
Jul 19 19:33:14 playapologies lighttpd[10300]: Unit openvpn-client@client.service could not be found.

RaspAP debug log generation complete.
```

Drizzt321, 2024-07-21:
Ok, interesting. After much trying many things, I've changed a few bits.

First, in the `/etc/raspap/hostapd/servicestartstop.sh` I've added the following line where it creates the `uap0` interface:

`ip address add 192.168.55.1/24 dev uap0`  (I modified the DHCP server settings to use 192.168.55.0). 

And in `/etc/network/interfaces` add 

```
auto uap0
iface uap0 inet static
        address 192.168.55.1/24
        pre-up iw dev wlan0 interface add uap0 type __ap
        pre-up ifconfig uap0 up
        up ip address add 192.168.55.1/24 dev uap0
```

After doing those 2 things, it boots, and then takes another minute or two, and then seems the `raspapd.service` runs again, and restarts the bits, hostapd, etc, and it's up and running and available to use.

billz, 2024-07-26:
As stated in the docs, this mode is experimental and completely unsupported

closed 2024-07-26 by billz

Drizzt321, 2024-07-26:
@billz you're referring to the AP-STA mode docs https://docs.raspap.com/ap-sta ? And just because it's marked as experimental means there's no bug fixes/support at all? I'd think the project would want to move towards it being fully supported eventually.

billz, 2024-07-28:
Won't happen. Many users failed to follow the steps and reported bugs like this one when AP-STA didn't work for them. For this reason, there's absolutely zero support.