RaspAP/raspap-webgui #1670: [Bug]: WiFi client doesn't handle double quotes in the password

number42net, 2024-10-05:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

Other

### Other software or services running with RaspAP?

No other software

### Bug description

WiFi client configuration in the web interface doesn't handle double quotes in the password. 

When entering a password with a double quote, the text after the double quote disappear after clicking "Update". When using double quotes around the password, the double quotes disappear after clicking "Update".

### Steps to reproduce

Set a password containing double quotes, click "Update" and then view the password again.

number42net, 2024-10-05:
I'm running RaspAP v3.1.8

closed 2024-10-06 by billz

billz, 2024-10-06:
@number42net thanks for reporting