RaspAP/raspap-webgui #1683: [Bug]: Eth0 enable fallback to static option not working in the qui

assachs, 2024-10-27:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

No other software

### Bug description

I want to achieve the following:
- if eth0 is connected to a network with a dhcp server, the ip-address provided by this server should be used
- if eth0 is connected to a network without a dhcp server, a fallback ip-address should be used and a dhcp-server should be startet

At the moment I am failing on the gui.
Under DHCP Server i select eth0,  "DHCP" and "Enable fallback to static option". But i can't insert the static parameters, because the fields are desabled.
Also "DHCP Options" are not enabled.

I'm not sure if "DHCP Options" should work, but at least i would expect, that the "static ip options" should be enabled.

### Steps to reproduce

Go to DHCP-Server
select eth0
select dhcp
enable "enable fallback to static option"
-> static ip options are disabled

billz, 2024-10-28:
A network interface can either have a static IP address or be assigned one by DHCP. If you select 'DHCP' in the UI, the fallback to static option becomes available. You may then configure a static IP:

[image]

The DHCP fields are disabled because a static IP address is required for this option. This is clearly indicated in the UI.

[image]

assachs, 2024-10-28:
To fallback:
i can't edit the IP field if dhcp and fallback is selected.

[image]

To DHCP-Server (maybe this is not a bug but a feature request):
if i edit the configuration files manually, i can add a dhcp server if the fallback ip is used. But this cannot be done with the gui.

(as far as i know, dnsmasq is only  responding to dhcp requests if the interface is part of the network of the range. That means it's only responding if the fallback ip is used (or if the ip address received by dhcp is in the same network as the fallback ip)

billz, 2024-10-28:
I see what you mean. The static IP fields should be enabled when the fallback option is selected.

PR #1684 (2024-10-28): Add event listener to chkfallback, enable/disable static IP fields
Resolves #1683
closed 2024-10-28 by billz