RaspAP/raspap-webgui #1688: Use pi-gen-action to build 32- and 64-bit RaspAP images

billz, 2024-10-31:
## Is your feature request related to a problem? 
Many users have requested a 32- or 64-bit Lite (bookworm) SD card image with a RaspAP installation. A pre-built image achieves two goals:

1. Faster setups without first following the [project perquisites](https://docs.raspap.com/#quick-start) and executing the installer.
2. Eliminates potential conflicts and troubleshooting when installing RaspAP on a non-clean OS.

## Describe the solution you'd like
The [pi-gen-action](https://github.com/marketplace/actions/pi-gen-action) wraps [pi-gen](https://github.com/RPi-Distro/pi-gen) to make it easily accessible in GitHub workflows. Generated images can subsequently be used in workflows, e.g. uploaded as build artifacts or further processed. `pi-gen` is the official tool to generate Raspberry Pi OS images that can be flashed on SD cards. A custom user stage may be defined to execute RaspAP's [quick installer](https://docs.raspap.com/quick/) with the following optional flags enabled (1 = enabled):

```
-o, --openvpn <flag>
-s, --rest, --restapi <flag>
-a, --adblock <flag>
-w, --wireguard <flag>
-g, --tcp-bbr <flag>

example: curl -sL https://install.raspap.com | bash -s -- --yes --openvpn 1 --restapi 1 --adblock 1 --wireguard 1 --tcp-bbr 1
```
After booting the OS with RaspAP, the user may then opt to [disable any of these components](https://docs.raspap.com/defaults/#managing-config-values) that aren't needed.

The build will be triggered whenever the master branch of the `/RaspAP/raspap-webgui` repository is updated. This will be done with a repository dispatch event and triggered by a webhook. Generated images will be [uploaded as artifacts](https://github.com/marketplace/actions/pi-gen-action#upload-final-image-as-artifact) that are accessible to end users.

## Describe alternatives you've considered
While not a strict requirement, `pi-gen-action` appears to provide a beaten path to creating the desired GitHub workflow.

## Additional context
Refer to the [pi-gen](https://github.com/RPi-Distro/pi-gen) repository for detailed information on the scripts and its usage.

billz, 2024-10-31:
@NL-TCH @glaszig possibly of interest :)

NL-TCH, 2024-10-31:
/attempt #1688

Dhanus3133, 2024-10-31:
/attempt #1688

sushantshah-dev, 2024-12-19:
Is the issue still looking for solutions?

BenraouaneSoufiane, 2025-01-27:
/attempt #1688

closed 2025-01-30 by billz