RaspAP/raspap-webgui #1713: [Bug]: Config Parser doesn't handle boolean flags (name-only options) properly

meandthemachine, 2024-12-16:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

No other software

### Bug description

In `functions.php` there is a function called `ParseConfig` that - among other files - is used to parse the dnsmasq conf file(s).

I guess that there is some code missing in this function that handles name-only boolean flags (options that have a name only and  don't contain an equal sign nor a value).
Below is version 3.2.2 of this function and I've added what I think is missing between the dashed `====` lines:
```
function ParseConfig($arrConfig, $wg = false)
{
    $config = array();
    foreach ($arrConfig as $line) {
        $line = trim($line);
        if ($line == "" || $line[0] == "#") {
            if ($wg) {
                $config[$option] = null;
                continue;
            } else {
                continue;
            }
        }

        if (strpos($line, "=") !== false) {
            list($option, $value) = array_map("trim", explode("=", $line, 2));
        } 
        // === HERE SOMETHING LIKE THIS SEEMS TO BE MISSING ================
        else {
            $option = $line; // Is already trimmed.
            $value = "";
        }
        // =====================================================

        if (empty($config[$option])) {
            $config[$option] = $value ?: true;
        } else {
            if (!is_array($config[$option])) {
                $config[$option] = [ $config[$option] ];
            }
            $config[$option][] = $value;
        }
    }
    return $config;
}
```

Because something like this is missing, defective behaviour can be observed. In my case, I have set Quad9 as upstream DNS and also specified some static leases, which leads to a dnsmasq conf file similar to this:

    # RaspAP wlan0 configuration
    interface=wlan0
    dhcp-range=10.3.141.50,10.3.141.70,255.255.255.0,12h
    dhcp-host=a8:be:57:a0:50:4d,10.3.141.100 # Client 1
    dhcp-host=9f:a0:93:8e:95:b7,10.3.141.101 # Client 2
    dhcp-host=ab:49:6f:8c:bb:d1,10.3.141.102 # Client 3
    no-resolv
    server=9.9.9.9

The parser doesn't respect/handle the line `no-resolv`. As a result, the for-loop-iteration that should handle it will instead reuse the - in this example - third static lease declaration of the previous iteration. This lease will then be listed twice in the UI and prevent from saving any other DHCP changes because of duplicate MAC addresses.

Of course, this bug also prevents other name-only options (like the logging flags) from being read from the conf files properly.
All those flags are missing in the resulting array of `ParseConfig`.

However, not all of the missing values result in an out-of-sync UI. For one some flags are directly grepped out of the pure CLI output (like ` $log_dhcp = (preg_grep('/log-dhcp/', $return))`), but I also guess that there is some JavaScript code in place that kind of heals the UI although flags have not been read/parsed properly (for example the missing `no-resolv` flag).

Please let me know if you need further information.
Thank you very much.

### Steps to reproduce

* Add upstream DNS in DHCP Advanced tab.
* Add static leases.

billz, 2024-12-16:
Sounds like a bug, will give it a look. Thanks for reporting

billz, 2024-12-17:
@meandthemachine your detailed bug report is much appreciated. As a small token, I'm glad to give you free access to Insiders as a way of saying 'thanks'. If this is interesting to you just say the word

PR #1714 (2024-12-17): Roundup of fixes for DHCP handling
Resolves #1713
Resolves several PHP warnings
closed 2024-12-17 by billz