RaspAP/raspap-webgui #1725: Adding WiFi 6

yoshideryu, 2025-01-08:
I am able to add wifi 6 to hostpad manually but if I change anything it's erased. 

Hoping for support add or possibly some custom fields to add whatever you want to the config

jp2692, 2025-01-09:
Hi, 

Theses changes help me in raspap-webgui with an option "Wifi 6 : On / off" in hotspot/basic tab

sudo :/var/www/html/templates/hostapd/nano basic.php

Replace the last line

</div>

by

  <div class="row">
    <div class="mb-3 col-md-6">
       <label for="cbxwifi6"><?php echo _("wifi 6 on/off"); ?>
              <select class="form-select" name="cbxwifi6" id="cbxwifi6">
                        <option value="ieee80211ax=0">off</option>
                        <option value="ieee80211ax=1"
                             <?php
                                  if (isset($arrConfig['ieee80211ax'])) {
                                        echo _('selected="selected"');
                                   }
                       ?>
                         >on</option>
                  </select>
           </label>
      </div>
  </div>
</div>

sudo nano /var/www/html/includes/hostapd.php

Find the lines  (towards the end of the file) :

    if ($_POST['wpa'] !== 'none') {
        $config.= 'wpa_passphrase='.$_POST['wpa_passphrase'].PHP_EOL;
    }

And before, add these lines :

   if ($_POST['cbxwifi6'] !== 'ieee80211ax=0') {
      $config.= 'ieee80211ax=1'.PHP_EOL;
       if ($_POST['hw_mode'] === 'ac') {
           $config.= 'he_oper_chwidth=1'.PHP_EOL;
           $config.= 'he_oper_centr_freq_seg0_idx=42'.PHP_EOL;
      }
   }

jp2692, 2025-01-09:
Sorry 
sudo :/var/www/html/templates/hostapd/nano basic.php

Replace the last line :
```

</div><!-- /.tab-pane | basic tab -->

```
by
```

  <div class="row">
    <div class="mb-3 col-md-6">
       <label for="cbxwifi6"><?php echo _("wifi 6 on/off"); ?>
              <select class="form-select" name="cbxwifi6" id="cbxwifi6">
                        <option value="ieee80211ax=0">off</option>
                        <option value="ieee80211ax=1"
                             <?php
                                  if (isset($arrConfig['ieee80211ax'])) {
                                        echo _('selected="selected"');
                                   }
                       ?>
                         >on</option>
                  </select>
           </label>
      </div>
  </div>
</div><!-- /.tab-pane | basic tab -->

```
sudo nano /var/www/html/includes/hostapd.php

Find the lines  (towards the end of the file) :
```

    if ($_POST['wpa'] !== 'none') {
        $config.= 'wpa_passphrase='.$_POST['wpa_passphrase'].PHP_EOL;
    }

```
And before, add these lines :
```

   if ($_POST['cbxwifi6'] !== 'ieee80211ax=0') {
      $config.= 'ieee80211ax=1'.PHP_EOL;
       if ($_POST['hw_mode'] === 'ac') {
           $config.= 'he_oper_chwidth=1'.PHP_EOL;
           $config.= 'he_oper_centr_freq_seg0_idx=42'.PHP_EOL;
      }
   }

```

yoshideryu, 2025-01-10:
Thank you I will try this soon and report back