RaspAP/raspap-webgui #1729: [Bug]: Installer silently fails on Rasberry Pi

dittert, 2025-01-12:
### Code of Conduct

- [X] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [X] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [X] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [X] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [X] I have searched this repository for existing issues.
- [X] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [X] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [X] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 5

### RaspAP version

3.1.3 (Latest)

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

dirk.dittert@googlemail.com

### Bug description

The quick installer silently fails on my Raspberry Pi.

### Steps to reproduce

Execute the quick installer:

```bash
curl -sL https://install.raspap.com | bash
```

### Additional context

The reason for this failure is that it fails to download the proper installer files. The following command fails in `_load_installer`:

```text
. . .
+ source=common
+ component=Install
+ wget -q https://raw.githubusercontent.com/RaspAP/raspap-webgui//installers/common.sh -O /tmp/raspap_common.sh
```

The `wget` command runs into a HTTP 404 error and fails silently. Thus, I am running into two errors:

1. The download URL is not properly constructed. It seems that the install script fails to determine that the branch `master` should be used.
2. The `wget` command fails silently in this case and does not report the error to the user

I have not fully analyzed what is going on in the installer script but I believe that the root cause could be in `_get_release` as the first `curl` to determine the latest release returns nothing.

### Relevant log output

```shell
There is no log output. The log file in /tmp is empty.
```

dittert, 2025-01-12:
Here is the full trace of what the installer is doing:

```
root@raspberrypi:~# ./raspbian.sh
+ _main
+ repo=RaspAP/raspap-webgui
+ repo_common=RaspAP/raspap-webgui
+ _parse_params
+ assume_yes=0
+ upgrade=0
+ update=0
+ ovpn_option=1
+ restapi_option=1
+ adblock_option=1
+ wg_option=1
+ insiders=0
+ minwrite=0
+ acctoken=
+ path=
+ :
+ case "${1-}" in
+ break
+ _setup_colors
+ ANSI_RED='\033[0;31m'
+ ANSI_GREEN='\033[0;32m'
+ ANSI_YELLOW='\033[0;33m'
+ ANSI_RASPBERRY='\033[0;35m'
+ ANSI_ERROR='\033[1;37;41m'
+ ANSI_RESET='\033[m'
+ _log_output
+ readonly LOGFILE_PATH=/tmp
+ LOGFILE_PATH=/tmp
+ '[' -f /tmp/raspap_install.log ']'
+ sudo rm /tmp/raspap_install.log
+ exec
+ exec
++ tee -i /tmp/raspap_install.log
+ _load_installer
+ _get_release
++ curl -s https://api.github.com/repos/RaspAP/raspap-webgui/releases/latest
++ grep -Po '"tag_name": "\K.*?(?=")'
+ readonly RASPAP_LATEST=
+ RASPAP_LATEST=
+ '[' 0 == 1 ']'
+ readonly RASPAP_RELEASE=
+ RASPAP_RELEASE=
+ '[' -z ']'
+ branch=
+ UPDATE_URL=https://raw.githubusercontent.com/RaspAP/raspap-webgui//
+ '[' '' = 1 ']'
+ '[' 0 = 1 ']'
+ '[' '' = 1 ']'
+ source=common
+ component=Install
+ wget -q https://raw.githubusercontent.com/RaspAP/raspap-webgui//installers/common.sh -O /tmp/raspap_common.sh
```

dittert, 2025-01-12:
Running the installer with `./raspbian.sh -b master` later fails with 

```text
Enable HttpOnly for session cookies (Recommended)? [Y/n]: Y
Php-cgi enabling session.cookie_httponly.
RaspAP Install: Cloning latest files from GitHub
Klone nach '/tmp/raspap-webgui'...
Schwerwiegend: konnte nicht auf 'https://github.com/RaspAP/raspap-webgui/' zugreifen: Failed to connect to github.com port 443 after 17 ms: Couldn't connect to server
[ ✘ error ]  Unable to download files from GitHub 
The installer cannot continue.
```

billz, 2025-01-12:
There's no issue with the installer. Your device doesn't have an internet connection 😉 

Try removing the `s` (silent) option. Refer to the curl manpage for more info.

```
$ curl -sL https://install.raspap.com | bash

 888888ba                              .d888888   888888ba
 88     8b                            d8     88   88     8b
a88aaaa8P' .d8888b. .d8888b. 88d888b. 88aaaaa88a a88aaaa8P
 88    8b. 88    88 Y8ooooo. 88    88 88     88   88
 88     88 88.  .88       88 88.  .88 88     88   88
 dP     dP  88888P8  88888P  88Y888P  88     88   dP
                             88
                             dP      version 3.2.4

The Quick Installer will guide you through a few easy steps
```

dittert, 2025-01-12:
Thank you for your quick response. I have looked into the issue for a couple more hours. It was caused by network problems (github.com was not resolved correctly in some circumstances). 

It can be closed unless there should be more error handling added to the installer script.

Sorry for the trouble.

billz, 2025-01-12:
Actually, I think you raise a good point. If the install loader is unable to reach github.com, it could notify the user that the device has connectivity issues. It's a better alternative than failing silently.

wolfiex, 2025-01-19:
I had a similar problem, which took longer than needed to debug as the installer did not provide adequate verbosicity on the second run attempt. 

( Once failed, the installer would not display any indication of trying to run again - suggesting, as it turns out incorrectly, that it would not. )
- I checked the cloning ability of git before the installer run and after, where an introduced change caused all clone attempts to fail. 
- This was solved using the code below. 

Solved by 
``` bash  

sudo nano /etc/resolv.conf

# insert the following line and save
# nameserver 8.8.8.8

sudo systemctl restart dnsmasq.service
```

OS: Armbian
SBC: Rock4C+

PR #1740 (2025-02-03): Adds _check_internet() to install loader
Includes a spinner to indicate progress.

https://github.com/user-attachments/assets/a89253d1-c28a-4be5-beef-7d2cf5d165b2

Resolves #1729
commit e61dac33 (2025-02-03): Add _check_internet() function. Resolves #1729
closed 2025-02-03 by billz