RaspAP/raspap-webgui #1746: [Bug]: #1740 completely breaking install now. connection verified

MSRS-Shop, 2025-02-03:
### Code of Conduct

- [x] I agree to follow this project's Code of Conduct.

### Issue reporting checklist

- [x] I have read and understand the [issue reporting policy](https://docs.raspap.com/issues/).
- [x] I observed this bug on a clean install of a [supported OS](https://docs.raspap.com/#compatible-operating-systems).
- [x] I have followed the [project prerequisites](https://docs.raspap.com/#quick-start).
- [x] I have searched this repository for existing issues.
- [x] I checked the [FAQ](https://docs.raspap.com/faq/) and [official documentation](https://docs.raspap.com/).
- [x] I am using an [external wireless adapter](https://docs.raspap.com/issues/#external-hardware).
- [x] I have generated a [RaspAP debug log](https://docs.raspap.com/ap-basics/#debug-log) and performed a [self-diagnosis](https://docs.raspap.com/ap-basics/#diagnosing-problems).

### Operating System

Raspberry Pi OS	(64-bit) Lite Bookworm

### Quick install or Manual setup?

Quick install

### Onboard wireless chipset or external adapter?

Onboard wireless chipset

### Hardware

Raspberry Pi 4 Model B

### RaspAP version

3.2.5 (Latest)

### Other software or services running with RaspAP?

No other software

### Contact details (optional)

rjdjohnston@gmail.com

### Bug description

The latest commit is blowing up install on all our devices when running ansible. 

I've tested commenting out the new method of checking internet and the install runs just fine. This line is the issue. its WAY to specific and if anything other then HTTP/2 200 is returned an exit code will not be set. 

The only work around now is a 4 task process to download the script, use regex to comment the _check_internet call in _main.

This is my temporary work around in ansible for anyone else having the same issue:

> - name: Download the RaspAP install script
  become: yes
  get_url:
    url: https://install.raspap.com
    dest: /tmp/raspap_install.sh
    mode: '0755'

- name: Comment out _check_internet call within the _main function
  become: yes
  lineinfile:
    path: /tmp/raspap_install.sh
    regexp: '^\s*_check_internet$'
    line: '    # _check_internet'
    insertafter: '^function _main\(\)'

- name: Execute the modified RaspAP install script
  become: yes
  shell: /tmp/raspap_install.sh --yes --openvpn 0 --adblock 0 --wireguard 0
  register: install_raspap
  until: install_raspap is succeeded
  retries: 3
  delay: 5

- name: Clean up the install script
  become: yes
  file:
    path: /tmp/raspap_install.sh
    state: absent

### Steps to reproduce

> - name: Install RaspAP
  become: yes
  shell: curl -sL https://install.raspap.com | bash -s -- --yes --openvpn 0 --adblock 0 --wireguard 0
  register: install_raspap
  until: install_raspap is succeeded
  retries: 3
  delay: 5

### Screenshots

> fatal: [msrs01]: FAILED! => {

    "attempts": 3,
    "changed": true,
    "cmd": "curl -sL https://install.raspap.com | bash -s -- --yes --openvpn 0 --adblock 0 --wireguard 0",
    "delta": "0:00:13.541009",
    "end": "2025-02-03 14:15:23.730095",
    "invocation": {
        "module_args": {
            "_raw_params": "curl -sL https://install.raspap.com | bash -s -- --yes --openvpn 0 --adblock 0 --wireguard 0",
            "_uses_shell": true,
            "argv": null,
            "chdir": null,
            "creates": null,
            "executable": null,
            "removes": null,
            "stdin": null,
            "stdin_add_newline": true,
            "strip_empty_ends": true
        }
    },
    "msg": "non-zero return code",
    "rc": 1,
    "start": "2025-02-03 14:15:10.189086",
    "stderr": "",
    "stderr_lines": [],
    "stdout": "\u001b[0;32mRaspAP Install: Checking internet connectivity...\u001b[m\n\u001b[?25l\r⠋\r⠙\r⠹\r⠸\r⠼\r⠴\r⠦\r⠧\r⠇\r⠏\r⠋\r⠙\r⠹\r⠸\r⠼\r⠴\r⠦\r⠧\r⠇\r⠏\r⠋\r⠙\r⠹\r⠸\r⠼\r⠴\r⠦\r⠧\r⠇\r⠏\r⠋\r⠙\r⠹\r⠸\r⠼\r⠴\r⠦\r⠧\r⠇\r⠏\r⠋\r⠙\r⠹\r⠸\r⠼\r⠴\r⠦\r⠧\r⠇\r⠏\r⠋\r⠙\r⠹\r⠸\r⠼\r⠴\r\u001b[?12l\u001b[?25h[\u001b[0;31m ✘ error \u001b[m] \u001b[1;37;41m No internet connection or unable to reach GitHub \u001b[m",
    "stdout_lines": [
        "\u001b[0;32mRaspAP Install: Checking internet connectivity...\u001b[m",
        "\u001b[?25l",
        "⠋",
        "⠙",
        "⠹",
        "⠸",
        "⠼",
        "⠴",
        "⠦",
        "⠧",
        "⠇",
        "⠏",
        "⠋",
        "⠙",
        "⠹",
        "⠸",
        "⠼",
        "⠴",
        "⠦",
        "⠧",
        "⠇",
        "⠏",
        "⠋",
        "⠙",
        "⠹",
        "⠸",
        "⠼",
        "⠴",
        "⠦",
        "⠧",
        "⠇",
        "⠏",
        "⠋",
        "⠙",
        "⠹",
        "⠸",
        "⠼",
        "⠴",
        "⠦",
        "⠧",
        "⠇",
        "⠏",
        "⠋",
        "⠙",
        "⠹",
        "⠸",
        "⠼",
        "⠴",
        "⠦",
        "⠧",
        "⠇",
        "⠏",
        "⠋",
        "⠙",
        "⠹",
        "⠸",
        "⠼",
        "⠴",
        "\u001b[?12l\u001b[?25h[\u001b[0;31m ✘ error \u001b[m] \u001b[1;37;41m No internet connection or unable to reach GitHub \u001b[m"
    ]
}

### Additional context

This 

### Relevant log output

```shell

```

billz, 2025-02-03:
> The latest commit 

If you're installing from the master (versus a release) branch, all bets are off. Nowhere is it stated that master is production ready.

> is blowing up install on all our devices when running ansible.

Whose ansible? Yours? Is the repo public?

MSRS-Shop, 2025-02-03:
Deploying from master means all bets are off? I could use a tag and when we update to latest I'd find the same error. Better to find earlier? I get what you're saying but thats not relevant to the issue.

The issue is the connection check code. I don't have a solution right now. Just ran into this issue while dug in on other things.

billz, 2025-02-03:
Installing from master in a production env is generally a pretty bad idea. This is what release versions are for and why the installer itself [uses tags](https://github.com/RaspAP/raspap-webgui/blob/aa49dcf32c83639b5d27d4b8256edc31cce4a3f9/installers/raspbian.sh#L246).

Users have raised valid concerns that the installer fails silently when the device has no connectivity. `_check_internet()` addresses this. If you can propose a better solution, or fix the existing one so it's compatible with your workflow PRs are always welcome.

rjdjohnston, 2025-02-03:
Ok. Setting curl timeout to 15 solves the issue for me. Created a pull request here:
https://github.com/RaspAP/raspap-webgui/pull/1747

closed 2025-02-05 by billz

billz, 2025-02-08:
@MSRS-Shop @rjdjohnston you can now skip the connectivity check with the `-k, --check` flag (`0` = [don't perform the check](https://docs.raspap.com/quick/#-k-check-flag)). This might be useful for your workflow 😉

rjdjohnston, 2025-02-09:
This is great. Thanks for the assist!